Open-Air-Foundation/firmware-one-openair #9: Missing return in periodicFetchData ?

chrissearle, 2021-09-11:
First up - the code compiles fine in the usual arduino IDE.

But - in platformio - where it builds with certain warnings enabled - in this case -Werror=return-type - we get:

```
AirGradient.cpp: In member function 'TMP_RH AirGradient::periodicFetchData()':
AirGradient.cpp:306:16: error: control reaches end of non-void function [-Werror=return-type]
  306 |     returnError(error);
```

I can fix it locally by just changing line 306 to:

```
return returnError(error);
```

I guess that the arduino IDE doesn't enable that warning.

But - it does look like a return on that line would be a good idea?

Not entirely sure - but thought I'd mention it.

Nightreaver, 2021-10-11:
I can confirm this issue - it should be fixed for platformio if it doesnt bother arduino IDE anyways

airgradienthq, 2021-12-08:
Thank you. It has been merged and is part of the latest release (1.4.0)

closed 2024-02-03 by airgradienthq