Open-Air-Foundation/firmware-one-openair #16: Why is CO2 RX and TX reversed in schematics

MingyaoLiu, 2021-11-27:
In the code the CO2_init is taking:

```
void AirGradient::CO2_Init(){
  CO2_Init(D4,D3);
}
```
 `CO2_Init(int rx_pin,int tx_pin){}`
```
void AirGradient::CO2_Init(int rx_pin,int tx_pin){
  CO2_Init(rx_pin,tx_pin,9600);
}
```

However in schematics the rx/tx is connected as such:
D4 --- TX
D3 --- RX

which seems like contradicting with the code where rx_pin is passed in D4, tx_pin is passed in D3.

Did I miss something?

jneuhauser, 2023-01-19:
If you connect two device over serial/uart it does look like this:
```
----------  RX <---------- TX  ----------
| Device |                     | Device |
----------  TX ----------> RX  ----------
```

That is, in words, the same physical connection is RX on one side and TX on the other. So it is not quite clear whether this connection/line is TX or RX. Usually the label right next to the device refers to the interface of the device.

MingyaoLiu, 2023-01-19:
Yup thanks for the info, I was a complete electric noob at that time, now I understand. Thanks!

closed 2024-02-03 by airgradienthq