Open-Air-Foundation/firmware-one-openair #30: Flashing via Web and Arduino yield different results (DIY Pro 4.2)

FunkeDope, 2023-06-23:
https://www.airgradient.com/open-airgradient/instructions/diy-pro-v42/

If you use Chrome to upload the firmware from the website, everything works fine. If you follow the instructions and flash the "Example" file provided (DIY_PRO_V4_2) there are several issues:

- The display is upside down.
- The device boot loops unless you press the hardware button to enter the config.

Does anyone have the actual sketch that was used to generate the web flash bin file?

brandonegg, 2023-06-30:
I'm facing the same issue as you. When flashing the example DIY_PRO_V4_2 I encounter a boot loop because it believes the button is pressed when it is not. I am not sure why there is descrepency between the version the web flasher uses and this repository. I have a fork of this repository [here](https://github.com/brandonegg/airgradient-influxdb) which was meant to add influxDB logging to the device. For a temporary fix of the problem, I just switched the setup() if statement to only enter the config menu if the button is low (I didn't both inverting this logic within the config logic since you can just configure those settings manually in the .ino file). This is odd because without a better understanding of the circuit board, the logic seems correct but it appears completing the button circuit presents a LOW signal. Perhaps board manager is misaligned with the D1 mini architecture they are using?

eprochasson, 2023-07-20:
It looks like the code for v4.2 has been overwritten with v3.7, as hinted by the top comment:

```cpp
/*
Important: This code is only for the DIY PRO PCB Version 3.7 that has a push button mounted.
```

The previous version https://github.com/airgradienthq/arduino/blob/cfc37d2d96e6a761740d672238f3bc2e08498a11/examples/DIY_PRO_V4_2/DIY_PRO_V4_2.ino seems to work fine, at least for me.

airgradienthq, 2024-02-03:
This has been fixed.

closed 2024-02-03 by airgradienthq