Open-Air-Foundation/firmware-one-openair #38: ONE V9 Dependencies List Incomplete

stevekuznetsov, 2023-12-27:
Attempting to compile and flash from source, the ONE_V9.ino seems to be missing some of the dependent libraries. I run into missing headers `PMS.h` and `s8_uart.h` - it looks like the latter is from [this library](https://www.arduino.cc/reference/en/libraries/s8_uart/) but I can't figure out what the former is from - I found [this library](https://www.arduino.cc/reference/en/libraries/pms-library/) but it has an incompatible `DATA` declaration:

```
/home/stevekuznetsov/Arduino/libraries/AirGradient_Air_Quality_Sensor/examples/ONE_V9/ONE_V9.ino: In function 'void updatePm()':
ONE_V9:290:26: error: 'struct PMS::DATA' has no member named 'PM_RAW_0_3'
       pm03PCount = data1.PM_RAW_0_3;
                          ^~~~~~~~~~
Multiple libraries were found for "WiFiClient.h"
 Used: /home/stevekuznetsov/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/WiFi
 Not used: /app/Arduino/libraries/WiFi
exit status 1
'struct PMS::DATA' has no member named 'PM_RAW_0_3'
```

stevekuznetsov, 2023-12-28:
Oh, interesting - looks like the `PMS::DATA` declaration that is being referred to is in your own `AirGradient.h` [here](https://github.com/airgradienthq/arduino/blob/f51203f754730f73ed6424b0b341e1a07af7dfd5/AirGradient.h#L191).

stevekuznetsov, 2023-12-28:
Trying to remove the `PMS.h` include and instead replace it with `AirGradient.h` doesn't compile:

```
/home/stevekuznetsov/Arduino/libraries/AirGradient_Air_Quality_Sensor/AirGradient.h:10:10: fatal error: SoftwareSerial.h: No such file or directory
 #include <SoftwareSerial.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board LOLIN C3 Mini
```

Doesn't look like there's a `SoftwareSerial` for the board?

stevekuznetsov, 2023-12-28:
Aaah ... perhaps [this](https://www.airgradient.com/blog/patching-pms-library-for-plantower-pms5003t/) is what I need? It is mentioned on the outdoor sensor guide but not on the ONE V9 one.

bluesky-ca, 2024-01-02:
Did you manage to compile it ?
   
In order to have the data stored locally I use dnsmasq and fake the IP for hw.airgradient.com to a local IP - I noticed the Arduino client does not respect TTL and I see dns queries every 10 sec for over 8K per day :)   Once I can compile it, will switch to an IP to avoid the DNS query.

Here are some other general suggestions to people that might be reading this - the WifiManager allows for custom config fields - it might be a good place to set some general options e.g. the LEDs.   

In addition, when the device is connected to a computer with a USB-C for firmware update it would be easy to have a serial menu that the user can interact with - the web flasher has a console option so it would be few lines of simple code - I found pressing the button on the unit difficult when looking at the screen - had to take it apart and hold my finger on the button to press it at the exact right moment (2 sec. window ) - have a console based setup menu would solve this.

stevekuznetsov, 2024-01-02:
@bluesky-ca yes! Sorry - I linked PR #35 here - that was what I ended up doing, with those steps it compiled. I contacted the maintainers to see if they will merge that PR.

bluesky-ca, 2024-01-02:
Searching for "PMS Library" find the original but it also matches one from airgradient - did you try that one to see if they patched the original one, or one needs to patch the original as they mention in their blog post:

Name: "AirGradient Air Quality Sensor"
  Author: AirGradient <support@airgradient.com>
  Maintainer: AirGradient <support@airgradient.com>
  Sentence: ESP8266 library for an air quality sensor featuring PM2.5, CO2, Temperature, TVOC and Humidity with OLED display.
  Paragraph: Air quality monitoring library supporting the Plantower PMS5003 particle sensor, the Senseair S8 CO2 sensor and the SHT3x/SHT4x sensor for humidity and temperature. Kits with all components including a nice enclosure are available in our online shop. You can also connect an OLED display or send the air quality data to the AirGradient platform or any other backend. Optionally you can connect the Sensirion SGP4x TVOC module from AirGradient.
  Website: https://www.airgradient.com/open-airgradient/instructions/
  Category: Sensors
  Architecture: *
  Types: Contributed
  Versions: [1.0.0, 1.3.1, 1.3.2, 1.3.3, 1.3.4, 1.3.5, 1.4.1, 1.4.2, 2.0.0, 2.0.2, 2.1.0, 2.2.0, 2.4.0, 2.4.1, 2.4.2, 2.4.3, 2.4.6, 2.4.7, 2.4.11, 2.4.12]

If this lib is the correct that would change the additional libs to:
"S8_UART"
"AirGradient Air Quality Sensor"

This would be simpler and not require patching the .h file.

stevekuznetsov, 2024-01-02:
You need the patched PMS Library as described [here](https://www.airgradient.com/blog/patching-pms-library-for-plantower-pms5003t/), as far as I could tell. Achim said they are in the middle of rewriting a bunch of this so I am not sure if that will be correct moving forward, it's just what worked for me.

bluesky-ca, 2024-01-02:
I was able to compile it ( thanks for all your work in getting the info together ) but only when I set the fqbn to esp32:esp32:XIAO_ESP32C3 - if I connect AirG. or any other C3 board I have and run:  arduino-cli board list for **_all of them_** I get:
        
/dev/ttyACM0 serial   Serial Port (USB) DFRobot Beetle ESP32-C3 esp32:esp32:dfrobot_beetle_esp32c3 esp32:esp32

Using fqbn with esp32:esp32:dfrobot_beetle_esp32c3 ( or esp32:esp32:esp32c3m1IKit  or esp32:esp32:esp32c3 ) fails to compile as there is no definition for Serial0 - should arduino-cli board list work and return a correct fqbn or is there a better way ?

Related to ONE_V9.ino I think the includes should be #include <PMS.h> #include <s8_uart.h> to make it obvious that the two files are not local to the project - compile works with "PMS.h" or <PMS.h>

lschloetterer, 2024-01-06:
@bluesky-ca 
> In order to have the data stored locally I use dnsmasq and fake the IP for hw.airgradient.com to a local IP - I noticed the Arduino client does not respect TTL and I see dns queries every 10 sec for over 8K per day :) Once I can compile it, will switch to an IP to avoid the DNS query.

What settings did you use for the DNS record? I have a local record in dnsmasq and it works just fine with the TTL

/etc/dnsmasq.d/10-local-domains.conf
```
host-record=pi.internal,192.168.0.10,900
cname=airgradient.internal,pi.internal,900
```

bluesky-ca, 2024-01-06:
I have been using a fake host method with addn-hosts=/etc/hosts.fake and min-cache-ttl=60 - I have very few hosts defined there as I use pi-hole to resolve most hosts, but looks like the fake host method is not applying TTL based on your comment - I will check it out - thanks for the tip.  

One thing about the addn-hosts is that you can use wildcard - it allows with one line to match all hosts of a domain, which can be useful ( I used this before setting up pi-hole ).

Based on your dns name using airgradient.internal I assume you recompiled the firmware and changed it from hw.airgradient.com - what value did you use for fqbn ?

lschloetterer, 2024-01-06:
Fqbn seems to be specific to the Arduino IDE. I'm using PlatformIO and set the board to lolin_c3_mini.
Maybe there is something similar in Arduino.

See here:
https://github.com/lschloetterer/airgradient-one-v9

bluesky-ca, 2024-01-06:
Thanks - lolin_c3_mini is there and I have used it to compile it, but if you look at the pin file I am not sure if the definitions match the schematic of the AirGradient - given that your setup works fine that implies that they defined all pins in the code and don't use default values - the problem comes when compiling code and assuming defaults e.g. default SPI pins when connected to a TFT display - I run into this problem and now create custom pin files for any boards I purchase from AliExpress :-)  

I only have arduino-cli installed as I am trying to duplicate the old portable setup from v1 - in v2 of the IDE I believe the portable feature is not working, so with the cli version I can have a custom arduino-cli.yaml with custom locations.

stevekuznetsov, 2024-01-08:
Closing this as the branch now has instructions.

closed 2024-01-08 by stevekuznetsov