Open-Air-Foundation/firmware-one-openair #39: IPv6 support

imp1sh, 2023-12-28:
Is the hardware / software combination able to handle IPv6? If so it would be nice to have it as an option in the Config secion…

imp1sh, 2024-03-01:
It seems there is no IPv6 support whatsoever, please consider this a feature request.

airgradienthq, 2024-03-29:
Unfortuntaley IP6 is currently not supported by the underlying libraries we use and it appears generally a problem for IoT devices like the ESP32.

imp1sh, 2024-03-29:
Thanks for clarification.
That's a bummer. What libraries are used? Can we open up a ticket upstream?
Especially IoT *should* be v6 capable I believe.

GoetzGoerisch, 2024-04-21:
IPv6 support is no problem for ESP32, lwIP and ESP-IDF support it.
BTW Tasmota has it, Esphome has it.

Which libs, that you are using does not support it?

sgryphon, 2024-04-21:
As mentioned above, all the underlying components have IPv6, so you can get it working by going to the lower level.

The higher level Arduino-ESP32 library will have IPv6 support in v3.0 (currently in RC): https://github.com/espressif/arduino-esp32/releases/tag/3.0.0-rc1