Open-Air-Foundation/firmware-one-openair #51: Narrowing Conversion in BoardDef.cpp

campfireman, 2024-02-12:
Hey guys,
I really appreciate the work you are doing! I'm in the midst of modifying the firmware of my AirGradient One. I came across an issue when compiling the 3.0.0 release:

`Air Quality Sensor/src/bsp/BoardDef.cpp:323:6: error: narrowing conversion of '-1' from 'int' to 'uint8_t' {aka 'unsigned char'}`

I'm on really thin ice here as I'm an absolute C++ noob, but I suspect the issue is that the `resetPin` in BoardDef.h is declared as `uint8_t` but assigned the value -1 in BoardDef.cpp?

I'm lacking the knowledge to make an actual fix, I was able to get around it with the compiler flag `-Wno-narrowing`.

pnt325, 2024-02-29:
Hi  @campfireman 

Thank for report this issue. I have update your issue at #62. I will soon release new version with this fix.

pnt325, 2024-02-29:
I'll close this issue. If any problems else feel free to open.

Thank,
Phat

closed 2024-02-29 by pnt325