Open-Air-Foundation/firmware-one-openair #55: No NOx value

oxivanisher, 2024-02-22:
I upgraded my AirGradient One V9 from 3.0.0 to 3.0.1 trough the [https://www.airgradient.com/documentation/one-v9/](airgradient.com/documentation/one-v9) page and realized that I now don't get a `noxIndex` value from the API from airgradient.com. Then I found the [https://www.airgradient.com/documentation/factory/](airgradient.com/documentation/factory/) page where 3.0.2 is available, but flashing this also did not fix the missing nox value. It is shown on the sensor display.

* Where should I post this? Since I did not change anything else than the firmware, I think this has to be the reason.
* Where is the correct place to find the actual latest (stable) firmware?

oxivanisher, 2024-02-29:
Currently I don't get the `nox_index` field from the airgradient API. But this might be because of the firmware version?
So my question remains: Where is the correct place to find the actual latest (stable) firmware?
Do I have to compile it myself for the bought AirGradient ONE V9?

pnt325, 2024-02-29:
> Currently I don't get the `nox_index` field from the airgradient API. But this might be because of the firmware version? So my question remains: Where is the correct place to find the actual latest (stable) firmware? Do I have to compile it myself for the bought AirGradient ONE V9?

Currently you are only get the lasted and stable firmware by compile the `ONE_I-9PSL.ino` example. On this example of version `3.0.4` you can get the data over `noxIndex`.

pnt325, 2024-03-01:
Here is newest firmware you can easy flash to your device: [3.0.5-beta](https://www.airgradient.com/documentation/firmwares/)

oxivanisher, 2024-03-09:
It is still confusing where and how new firmwares are linked for download, but on this url, https://www.airgradient.com/documentation/firmwares/ there is now even a 3.0.7 and it seems that my initial problem is solved, so closing this issue.

closed 2024-03-09 by oxivanisher