Open-Air-Foundation/firmware-one-openair #82: Server POST not needed when using Prometheus endpoint

tykling, 2024-03-13:
Hello,

Thank you for a great product.

The new Prometheus `/metrics` endpoint is great, so I no longer need to `POST` periodically to a server. But the code requires successful `POST` to the server URL or it turns on an error LED.

Would it be possible to make periodic server `POST` reporting configurable in the server config? Then people can disable it if they don't need it.

Thanks! :)

pnt325, 2024-03-14:
Hi @tykling 

You can find on code and comment this line:
```c
serverSchedule.run();
```

tykling, 2024-03-18:
Hello,

Thank you for that.

I know I can disabe it in the code, but I would love to be able to turn it off in the serverconfig. Any chance of adding this? 

The server POST url could be made configurable in the serverconfig, and if left blank it could disable server POST entirely (including the error logic).

Thoughts? Thank you! :)

pnt325, 2024-03-24:
Hi @tykling 
Thank for advice. 

Currently we don't support it from server configuration.
I will discuss with team about it.

Thanks
Phat

airgradienthq, 2024-03-29:
We plan to add this as part of the configuration of the monitor. So that a PUT can be set to toggle the communication with the AirGradient servers.

airgradienthq, 2024-07-10:
This has been addressed in the latest firmware release.

closed 2024-07-10 by airgradienthq