Open-Air-Foundation/firmware-one-openair #139: TVOC in display should be in all caps

MallocArray, 2024-05-18:
Like other sensor names, TVOC should be in all caps, not lower case.

It is an acronym and should be capitalized, as well as matching the formatting of all of the other displayed readings.

Also, I believe it should be VOC instead of TVOC, as that is what the manufacturer refers to it at:
https://sensirion.com/products/catalog/SGP40/

pnt325, 2024-07-08:
@airgradienthq 
[image]

MallocArray, 2024-07-09:
I like the capitals.

Still think the spacing of the columns could be improved. No reason for all of the blank space to the right of VOC and PM2.5 could still use some extra room.

This is what my ESPHome firmware looks like (Top line wasn't included due to my camera shutter being too fast)

[image]

airgradienthq, 2024-07-10:
@pnt325 Please also adjust the layout as @MallocArray suggests. Then make a new pull request.

pnt325, 2024-07-18:
[image]

I update the layout for 4 character value of CO2 and 3 character value of PM.

MallocArray, 2024-07-18:
I like it!

airgradienthq, 2024-07-18:
@pnt325 Perfect. Let's go with that.

pnt325, 2024-07-19:
@airgradienthq Please check and aprrove PR #195

closed 2024-07-19 by pnt325