Open-Air-Foundation/firmware-one-openair #140: PM2.5 display could be wider

MallocArray, 2024-05-18:
The PM2.5 window in the middle of the display looks oddly squished, especially since there is so much blank space to the far right of `tvoc:`

I also wonder if PM2.5 would properly handle 3 digits, as it looks to only have sufficient space to look good for up to 2 digits. 

What if it needed to display 288?  Feels like it may not fit within the bars, or at least would be very cramped looking.

MallocArray, 2024-05-18:
CO2 also could/should be wider.  I had it displaying an 8 as the last digit and it looked very uncentered in the column and nearly touching the bar.  Here is a picture after it changed to 9, and while this photo doesn't look at bad, in person, it certainly felt misaligned and like it was running off the side of the column
[image]

airgradienthq, 2024-07-18:
Closing thus due to duplicate.

closed 2024-07-18 by airgradienthq