Open-Air-Foundation/firmware-one-openair #141: Display went dark and LED at low brightness

MallocArray, 2024-05-18:
After flashing the latest 3.1.1 firmware, I had the display and LED working at normal levels, and now the display has gone completely off, and the LEDs are at a low brightness level

[image]

I assume this is some kind of night mode kicking in, but I haven't configured a night/sleep mode in the Dashboard, nor can I even find where that may be an option to enable or configure.

In my time zone it is only 8:00 pm and not when I would want the unit to be in a sleep mode, but the end user should

1. Opt-in for sleep mode
2. Be able to configure the sleep time window in the Dashboard

This is all assuming this isn't just a bug.

airgradienthq, 2024-05-18:
Yes it's the night mode. The default off is confusing and we will need to change. 
you can adjust the settings already in the dashboard. 

More information here:
https://forum.airgradient.com/t/firmware-3-1-1-released/2077/9

airgradienthq, 2024-07-10:
This has been addressed server side in the meantime.

closed 2024-07-10 by airgradienthq