Open-Air-Foundation/firmware-one-openair #142: Update ignored due to local unofficial changes

MallocArray, 2024-05-18:
Looking at logs after installing the 3.1.1 firmware, I see a message that I was not expecting:

```
[Configure] Info: Parse configure: {"country":"US","pmStandard":"ugm3","co2CalibrationRequested":false,"ledBarTestRequested":false,"ledBarMode":"co2","displayMode":"on","placeOpen":false,"abcDays":8,"tvocLearningOffset":12,"noxLearningOffset":12,"temperatureUnit":"c","targetFirmware":"3.1.1","ledBarBrightness":10,"displayBrightness":0,"model":"I-9PSL-DE"}
[Configure] Info: Parse configure success
[Configure] Info: Update ignored due to local unofficial changes
```

I haven't attempted to make any local changes with the API or anything, just flashed the firmware and then opened the logs.  

Is this message misleading or what would cause it to see local unofficial changes?

airgradienthq, 2024-05-18:
@MallocArray did you flash the firmware through our website flasher or compiled it locally and flashed it through Arduino or Platform IO?

MallocArray, 2024-05-18:
Web flash using Edge browser in Windows 11 over USB. I checked the box to Erase

https://www.airgradient.com/documentation/firmwares/

danielnitz, 2024-05-18:
I see the exact same thing, same procedure expect on MacOS and Chrome browser.

airgradienthq, 2024-05-18:
Tracks for reporting. We are looking into this.

airgradienthq, 2024-07-01:
This appears to be fixed in the latest firmware. Testes on 3.1.4.

closed 2024-07-01 by airgradienthq