Open-Air-Foundation/firmware-one-openair #145: Boot counter not working in Dashboard

MallocArray, 2024-05-18:
I'm not sure if this is most appropriate here or in the Forums, since it seems to be more related to the Dashboard page.

Going to the Hardware page I can see all of my devices and when they were last seen:
https://app.airgradient.com/settings/hardware

The "Current Boot Count" is not incrementing on my ONE that I recently upgraded to 3.1.1.  I certainly see in the logs that it is sending a value for "bootCount" that is incrementing, but the Dashboard only shows 0.
I suspect it may be an incorrect value name, since this link shows that it should be called `boot` and not `bootCount`
https://github.com/airgradienthq/arduino/blob/master/docs/local-server.md

MallocArray, 2024-05-19:
Confirmed that changing to `bootCount` does properly show the Current Boot Count in the Hardware page of the Dashboard.

airgradienthq, 2024-07-01:
Fixed