Open-Air-Foundation/firmware-one-openair #147: Open Air O-1PST logs showing various errors

MallocArray, 2024-05-19:
First boot of a device after clean install of the 3.1.1 firmware.  This was while I was connecting to the hotspot on my phone and configuring my Wifi information.

Note the `request handler not found` and `No more processes` errors

I did not have this unit setup in the Dashboard, so I also got some Error 400 but I don't think that is related to the errors mentioned, unless it is due to not being able to retrieve configuration from the Dashboard.

```
*wm:-2 networks found
*wm:2 networks found
[ 61179][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 61185][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 61240][E][WebServer.cpp:649] _handleRequest(): request handler not found
*wm:Connecting to NEW AP: NachoWifi
*wm:Connect to new AP [SUCCESS] 
*wm:Got IP Address: 
*wm:192.168.2.244 
[ 69910][E][WebServer.cpp:649] _handleRequest(): request handler not found
[WifiConnector] Info: Setting postToAirGradient set from True to True successful
[Configure] Info: postDataToAirGradient: Ignored set to 1
[LocalServer] Info: Init: airgradient_xxxxxxx.local
Mqtt uri is empty
Connect to MQTT broker failed
[ApiClient] Info: Post uri: http://hw.airgradient.com/sensors/airgradient:xxxxxxx/measures
[ApiClient] Info: Post data: {"wifi":-57,"boot":0}
[ 73868][E][WiFiClient.cpp:517] flush(): fail on fd 49, errno: 11, "No more processes"
[ 88502][E][WiFiClient.cpp:517] flush(): fail on fd 49, errno: 11, "No more processes"
[104879][E][WiFiClient.cpp:517] flush(): fail on fd 49, errno: 11, "No more processes"
[ApiClient] Info: Post uri: http://hw.airgradient.com/sensors/airgradient:84fce60becfc/measures
[ApiClient] Info: Post data: {"wifi":-57,"pm01":-1,"pm02":-1,"pm10":-1,"pm003Count":-1,"atmp":-1000.99,"rhum":-1,"bootCount":0}
[105800][E][WiFiClient.cpp:517] flush(): fail on fd 49, errno: 11, "No more processes"

pnt325, 2024-05-24:
Hi @MallocArray 

It's look like the issue from WiFiManager when you're provisioning WiFi connection. How often you're a lot of log failed message like that. Cause on the first time when you're open web portal, device also scanning WiFi SSID thatn make the web portal show failed message.

MallocArray, 2024-05-24:
The first few lines would relate to me using the hotspot and scanning for SSIDs, but later after the Post uri line would have been in normal operation after the hotspot had shut down and was using the full Wifi in the house.

I saw the "No more processes" repeatedly while sitting idle.

Again, I didn't have this device setup in the Dashboard, so I expect an Error 400, but don't think that would cause that message, would it?  If it would, a better error message would be an improved user experience

pnt325, 2024-05-24:
So on WiFiManager portal it's has the checkbox `Prevent Connection to Airgradient Server`. If you're do not setup device on dashboard please un-check it.
[image]

airgradienthq, 2024-09-25:
@MallocArray I believe this has been addressed for quite some time now. Are you ok if we close this issue?

closed 2024-09-25 by MallocArray