Open-Air-Foundation/firmware-one-openair #151: AirGradient ONE (indoor) not reconnecting to wifi after outage

groner, 2024-05-24:
I have two indoor AirGradient ONE's running the 3.1.1 firmware.

Today, following a brief wifi interruption one of them reconnected and the other one did not.

I don't have it connected to a computer, so I don't have any log output to share.

pnt325, 2024-05-24:
Hi @groner 

Without the log message it's difficult to identify the issues. 

How about your network traffic? Cause if your WiFi router has limit is reach limit of connection device at time it's only allow the device has high performance WiFi like mobile phone, laptop, or near by device.

groner, 2024-05-24:
Reviewing some logs on the wifi access point here, I can see that my router went down (ether1 link down) but the AP stayed up. Perhaps an error in a higher network layer leads to the airgradient disassociating from the wifi network?

```
 05-22 19:59:26 wireless,info EC:DA:3B:xx:xx:xx@LOCAL-2GHZ connected, signal strength -32
 05-23 19:54:40 interface,info ether1 link down
 05-23 19:55:05 wireless,debug EC:DA:3B:xx:xx:xx@LOCAL-2GHZ reauthenticating
 05-23 19:55:06 wireless,debug EC:DA:3B:xx:xx:xx@LOCAL-2GHZ reauthenticating
 05-23 19:55:07 wireless,info EC:DA:3B:xx:xx:xx@LOCAL-2GHZ disconnected, SA Query timeout, signal strength -32
 05-23 19:55:09 wireless,info EC:DA:3B:xx:xx:xx@LOCAL-2GHZ connected, signal strength -31
 05-23 19:56:26 wireless,info EC:DA:3B:xx:xx:xx@LOCAL-2GHZ disconnected, connection lost, signal strength -30
 05-23 20:00:00 interface,info ether1 link up (speed 2.5G, full duplex)
 05-23 20:00:33 interface,info ether1 link down
 05-23 20:00:44 interface,info ether1 link up (speed 2.5G, full duplex)
 05-23 21:28:53 wireless,info EC:DA:3B:xx:xx:xx@LOCAL-2GHZ connected, signal strength -34
```

pnt325, 2024-05-24:
```
 Perhaps an error in a higher network layer leads to the airgradient disassociating from the wifi network?
```

Maybe, to verify that case you can turn off or disconnect some device from network then turn arigradient on to see what happen. I has testing again device reconnect on my network router it's worked as well on `3.1.1`

airgradienthq, 2024-07-01:
This should be fixed in 3.1.4