Open-Air-Foundation/firmware-one-openair #160: [docs] instructions to compile

hestiahacker, 2024-06-02:
# Problem
Attempting to compile in the Arduino IDE results in the following error:

```
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:40:10: fatal error: AirGradient.h: No such file or directory
   40 | #include "AirGradient.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: AirGradient.h: No such file or directory
```

The tactical problem issue here is that some included files are in examples/OneOpenAir, while others are in src, and Arduino IDE is not looking in all these places.

However, the core problem is I can not find instructions on how to configure the Arduino IDE to compile. Since I expect others can compile the code, I expect this is just a setup problem and not a broken codebase.

# Steps to reproduce
1. Set up Arduino IDE to compile for the ESP32 board using the [instructions on the blog](https://www.airgradient.com/blog/install-arduino-c3-mini/)
2. Install the "serial" library for Python (e.g. `sudo apt install python3-serial`)
3. Checkout this repo and open `./examples/OneOpenAir/OneOpenAir.ino` in the Arduino IDE
4. Click the verify button (check mark button in the upper left of the Arduino IDE) and see the posted error above

# Attempts at resolving this
I tried a number of things to resolve this.

- Make a symlink pointing to `./examples/OneOpenAir/OneOpenAir.ino`, put it in `./src/`, and open and compile that (so the file is next to the .h file we want to include)
- Copy `./examples/OneOpenAir/OneOpenAir.ino` into `./src/`, and open and compile that (so the file is next to the .h file we want to include)
- Make a symlink to `./src/Libraries` named `./examples/OneOpenAir/libraries` so the sketch directory has a libraries folder, as described in the [arduino documentation](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries/)
- File -> Preference -> Change Sketchbook location to the root git repo, `./src`, or `./examples/OneOpenAir` to get it to find the libraries directory
- copy OneOpenAir.ino into the `./src` directory, rename `./src` to `OneOpenAir`, open OneOpenAir.ino in the Arduino IDE and attempt to compile it (this was tried because Arduino insists on having the name of the directory match the name of the .ino file)

All of the above gave same error message except the last one, which give the error message below:

```
/home/user/air_gradient_open_air_firmware/OneOpenAir/OneOpenAir.ino:41:10: fatal error: OtaHandler.h: No such file or directory
   41 | #include "OtaHandler.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: OtaHandler.h: No such file or directory
```

Symlinking to ../examples/OneOpenAir/OtaHandler.h, ../examples/OneOpenAir/LocalServer.h and ../examples/OneOpenAir/OpenMetrics.h works around the "No such file" errors, but then exposes the more serious compilation errors below:

```
home/user/air_gradient_open_air_firmware/OneOpenAir/AgConfigure.cpp: In member function 'bool Configuration::parse(String, bool)':
/home/user/air_gradient_open_air_firmware/OneOpenAir/AgConfigure.cpp:277:45: error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  277 |     if (jconfig[jprop_configurationControl] ==
In file included from /home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/Arduino.h:192,
                 from /home/user/air_gradient_open_air_firmware/OneOpenAir/Main/PrintLog.h:4,
                 from /home/user/air_gradient_open_air_firmware/OneOpenAir/AgConfigure.h:5,
                 from /home/user/air_gradient_open_air_firmware/OneOpenAir/AgConfigure.cpp:1:
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.h:211:8: note: candidate: 'bool String::operator==(const String&) const' (reversed)
  211 |   bool operator==(const String &rhs) const {
      |        ^~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.h:214:8: note: candidate: 'bool String::operator==(const char*) const' (reversed)
  214 |   bool operator==(const char *cstr) const {
      |        ^~~~~~~~
In file included from /home/user/air_gradient_open_air_firmware/OneOpenAir/Libraries/Arduino_JSON/src/JSON.h:25,
                 from /home/user/air_gradient_open_air_firmware/OneOpenAir/Libraries/Arduino_JSON/src/Arduino_JSON.h:23,
                 from /home/user/air_gradient_open_air_firmware/OneOpenAir/AgConfigure.cpp:2:
/home/user/air_gradient_open_air_firmware/OneOpenAir/Libraries/Arduino_JSON/src/JSONVar.h:85:8: note: candidate: 'bool JSONVar::operator==(const JSONVar&) const'
   85 |   bool operator==(const JSONVar& v) const;
      |        ^~~~~~~~
/home/user/air_gradient_open_air_firmware/OneOpenAir/AgConfigure.cpp:286:45: error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  286 |     if (jconfig[jprop_configurationControl] ==
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.h:211:8: note: candidate: 'bool String::operator==(const String&) const' (reversed)
  211 |   bool operator==(const String &rhs) const {
      |        ^~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.h:214:8: note: candidate: 'bool String::operator==(const char*) const' (reversed)
  214 |   bool operator==(const char *cstr) const {
      |        ^~~~~~~~
/home/user/air_gradient_open_air_firmware/OneOpenAir/Libraries/Arduino_JSON/src/JSONVar.h:85:8: note: candidate: 'bool JSONVar::operator==(const JSONVar&) const'
   85 |   bool operator==(const JSONVar& v) const;
      |        ^~~~~~~~
/home/user/air_gradient_open_air_firmware/OneOpenAir/AgApiClient.cpp: In member function 'bool AgApiClient::postToServer(String)':
/home/user/air_gradient_open_air_firmware/OneOpenAir/AgApiClient.cpp:108:7: error: 'WiFi' was not declared in this scope
  108 |   if (WiFi.isConnected() == false) {
      |       ^~~~
/home/user/air_gradient_open_air_firmware/OneOpenAir/AgApiClient.cpp:118:3: error: 'WiFiClient' was not declared in this scope
  118 |   WiFiClient wifiClient;
      |   ^~~~~~~~~~
/home/user/air_gradient_open_air_firmware/OneOpenAir/AgApiClient.cpp:120:20: error: 'wifiClient' was not declared in this scope
  120 |   if (client.begin(wifiClient, uri.c_str()) == false) {
      |                    ^~~~~~~~~~

exit status 1

Compilation error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
```

hestiahacker, 2024-06-02:
Update: I found the [Flashing Of The AirGradient Firmware](https://www.airgradient.com/documentation/one-v9/#flashing-of-the-airgradient-firmware) instructions. This resulted in a 'class Configuration' has no member named 'isOfflineMode' error message.

This is with version 3.1.0-beta.1 of the AirGradient library and commit 2e62abe2d7d8a3c549f04b78bfcff8c745ce9af0 of this repo.

The Flashing instructions say to install additional libraries listed on the top of the example code, but the example code does not specify any additional libraries to install.

```
home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino: In function 'void setup()':
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:184:25: error: 'class Configuration' has no member named 'isOfflineMode'
  184 |           configuration.isOfflineMode() ? "online mode" : "offline mode", "");
      |                         ^~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:188:25: error: 'class Configuration' has no member named 'setOfflineMode'
  188 |           configuration.setOfflineMode(!configuration.isOfflineMode());
      |                         ^~~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:188:55: error: 'class Configuration' has no member named 'isOfflineMode'
  188 |           configuration.setOfflineMode(!configuration.isOfflineMode());
      |                                                       ^~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:192:29: error: 'class Configuration' has no member named 'isOfflineMode'
  192 |               configuration.isOfflineMode() ? " = True" : "  = False", "");
      |                             ^~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:201:38: error: 'class Configuration' has no member named 'isOfflineMode'
  201 |       connectToWifi = !configuration.isOfflineMode();
      |                                      ^~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:203:21: error: 'class Configuration' has no member named 'setOfflineModeWithoutSave'
  203 |       configuration.setOfflineModeWithoutSave(true);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:225:24: error: 'class Measurements' has no member named 'otaBootCount'; did you mean 'bootCount'?
  225 |           measurements.otaBootCount = 0;
      |                        ^~~~~~~~~~~~
      |                        bootCount
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:232:27: error: 'class AgApiClient' has no member named 'isNotAvailableOnDashboard'
  232 |             if (apiClient.isNotAvailableOnDashboard()) {
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:244:21: error: 'class OledDisplay' has no member named 'showRebooting'
  244 |         oledDisplay.showRebooting();
      |                     ^~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:250:21: error: 'class WifiConnector' has no member named 'hasConfigurated'
  250 |   if (wifiConnector.hasConfigurated() == false) {
      |                     ^~~~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:252:19: error: 'class Configuration' has no member named 'setOfflineModeWithoutSave'
  252 |     configuration.setOfflineModeWithoutSave(true);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:260:66: error: 'class Configuration' has no member named 'getDisplayBrightness'
  260 |     Serial.println("Display brightness: " + String(configuration.getDisplayBrightness()));
      |                                                                  ^~~~~~~~~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:261:17: error: 'class OledDisplay' has no member named 'setBrightness'
  261 |     oledDisplay.setBrightness(configuration.getDisplayBrightness());
      |                 ^~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:261:45: error: 'class Configuration' has no member named 'getDisplayBrightness'
  261 |     oledDisplay.setBrightness(configuration.getDisplayBrightness());
      |                                             ^~~~~~~~~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino: In function 'void loop()':
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:302:21: error: 'class Configuration' has no member named 'isOfflineMode'
  302 |   if (configuration.isOfflineMode() ||
      |                     ^~~~~~~~~~~~~
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino: In function 'void ledBarEnabledUpdate()':
/home/user/air_gradient_open_air_firmware/examples/OneOpenAir/OneOpenAir.ino:465:36: error: 'class Configuration' has no member named 'getLedBarBrightness'
  465 |     int brightness = configuration.getLedBarBrightness();
      |                                    ^~~~~~~~~~~~~~~~~~~
[... cut ...]
                 from /home/user/air_gradient_open_air_firmware/examples/OneOpenAir/LocalServer.cpp:1:
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.h:67:10: note: candidate: 'String Configuration::toString()'
   67 |   String toString(void);
      |          ^~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.h:67:10: note:   candidate expects 0 arguments, 1 provided

exit status 1

Compilation error: 'class Configuration' has no member named 'isOfflineMode'
```

hestiahacker, 2024-06-02:
After some additional research, it appears the commit which added isOfflineMode (commit da6326db0fded87110163d7fc93f200afe6e202d) is only in tags 3.1.1 and 3.1.2 (as seen with `git tag -l --contains da6326db0fded87110163d7fc93f200afe6e202d`).

The latest version available in the Arduino Library is 3.1.0-beta.1, which is why I'm running into this issue. If the Arduino library can not be automatically released when [a firmware version is released](https://forum.airgradient.com/t/firmware-3-1-1-released/2077), perhaps we can add instructions on how to configure the Arduino IDE use the local git checkout instead of having to install the library with the library manager to ensure this is resolved going forward?

pnt325, 2024-06-08:
Hi @hestiahacker 

You can download firmware on branch [master](https://github.com/airgradienthq/arduino/tree/master) for office release. For newest support function and bugfix on branch [develop](https://github.com/airgradienthq/arduino/tree/develop).

The Arduino IDE only get the firmware for office release after 1-2 days.

hestiahacker, 2024-06-08:
The stacktrace from [my previous comment](https://github.com/airgradienthq/arduino/issues/160#issuecomment-2143715860) was from trying to compile the firmware on the master branch.

Since the, version 3.1.3 of the library has been released and is available in the published libraries, however the master branch still does not compile, this time with a different error because the otaBootCount member doesn't exist.

The best way to solve this problem is to have the OneOpenAir.ino sketch use the local files to compile instead of requiring a library to be installed. As you seem to have indicated, the library available via the Arduino IDE will always lag behind.

My goal in opening this ticket is to ensure a new developer can always compile at least the master branch, and ideally also the develop branch as well. I have started creating this documentation, but since I am still unable to compile, clearly the documentation is not yet ready to be submitted to be merged in.

pnt325, 2024-06-08:
Hi @hestiahacker 
All build is worked without any problem and `otaBootCount` is removed from code since `3.1.3`

Please follow instruction begin of file:
https://github.com/airgradienthq/arduino/blob/8a83e408d3c036b089b63c053a11364b41a26092/examples/OneOpenAir/OneOpenAir.ino#L12-L27

If it's not worked, please remove the library on your PC then install again.

abstractpoint, 2024-06-15:
I'm also having compile issues, and I have followed these instructions above.
```
/Users/temp/Documents/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp: In member function 'bool Configuration::parse(String, bool)':
/Users/temp/Documents/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:277:45: error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  277 |     if (jconfig[jprop_configurationControl] ==
In file included from /Users/temp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.1/cores/esp32/Arduino.h:192,
                 from /Users/temp/Documents/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Main/PrintLog.h:4,
                 from /Users/temp/Documents/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.h:5,
                 from /Users/temp/Documents/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:1:
/Users/temp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.1/cores/esp32/WString.h:211:8: note: candidate: 'bool String::operator==(const String&) const' (reversed)
  211 |   bool operator==(const String &rhs) const {
      |        ^~~~~~~~
...
```

pnt325, 2024-06-16:
> I'm also having compile issues, and I have followed these instructions above. 3.1.3 firmware and Library. Docs for compiling is a great idea.
> 
> ```
> /Users/temp/Repositories/2024/arduino/examples/OneOpenAir/OneOpenAir.ino: In function 'void oneIndoorInit()':
> /Users/temp/Repositories/2024/arduino/examples/OneOpenAir/OneOpenAir.ino:722:25: error: 'Serial0' was not declared in this scope
>    if (ag->pms5003.begin(Serial0) == false) {
>                          ^~~~~~~
> /Users/temp/Repositories/2024/arduino/examples/OneOpenAir/OneOpenAir.ino:722:25: note: suggested alternative: 'Serial1'
>    if (ag->pms5003.begin(Serial0) == false) {
>                          ^~~~~~~
> ...
> ```

You're missed `Enable CDC On Boot`

hestiahacker, 2024-06-17:
I've added [documentation on compiling](https://github.com/hestiahacker/arduino/blob/master/docs/compiling.md) based on what I've collected and what was said in this thread, but this project still doesn't compile.

I hope someone will be able to follow the development setup instructions and reproduce this issue so we can get to the bottom of it and get it resolved.

# Still unable to compile
I have gone through the instructions for setting up a development environment again on a new machine using AirGradient Air Quality Sensor library version 3.1.3 and I am still unable to compile the firmware. I am on the "master" branch, and more specifically: on commit 3b6859f483f494d0556d36acdb95bfc3d7957f08.

```
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp: In member function 'bool Configuration::parse(String, bool)':
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:277:45: error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  277 |     if (jconfig[jprop_configurationControl] ==
In file included from /home/user/.arduino15/packages/esp32/hardware/esp32/3.0.1/cores/esp32/Arduino.h:192,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Main/PrintLog.h:4,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.h:5,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:1:
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.1/cores/esp32/WString.h:211:8: note: candidate: 'bool String::operator==(const String&) const' (reversed)
  211 |   bool operator==(const String &rhs) const {
      |        ^~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.1/cores/esp32/WString.h:214:8: note: candidate: 'bool String::operator==(const char*) const' (reversed)
  214 |   bool operator==(const char *cstr) const {
      |        ^~~~~~~~
In file included from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/JSON.h:25,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/Arduino_JSON.h:23,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:2:
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/JSONVar.h:85:8: note: candidate: 'bool JSONVar::operator==(const JSONVar&) const'
   85 |   bool operator==(const JSONVar& v) const;
      |        ^~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:286:45: error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  286 |     if (jconfig[jprop_configurationControl] ==
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.1/cores/esp32/WString.h:211:8: note: candidate: 'bool String::operator==(const String&) const' (reversed)
  211 |   bool operator==(const String &rhs) const {
      |        ^~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.1/cores/esp32/WString.h:214:8: note: candidate: 'bool String::operator==(const char*) const' (reversed)
  214 |   bool operator==(const char *cstr) const {
      |        ^~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/JSONVar.h:85:8: note: candidate: 'bool JSONVar::operator==(const JSONVar&) const'
   85 |   bool operator==(const JSONVar& v) const;
      |        ^~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgApiClient.cpp: In member function 'bool AgApiClient::postToServer(String)':
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgApiClient.cpp:108:7: error: 'WiFi' was not declared in this scope
  108 |   if (WiFi.isConnected() == false) {
      |       ^~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgApiClient.cpp:118:3: error: 'WiFiClient' was not declared in this scope
  118 |   WiFiClient wifiClient;
      |   ^~~~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgApiClient.cpp:120:20: error: 'wifiClient' was not declared in this scope
  120 |   if (client.begin(wifiClient, uri.c_str()) == false) {
      |                    ^~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

If we look at there errors, we can see they match up with the source code in this repo, so this is not a matter of having outdated code.

- [src/AgConfigure.cpp:277:45](https://github.com/airgradienthq/arduino/blob/master/src/AgConfigure.cpp#L277): error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  - jconfig is a [JSONVar object](https://github.com/airgradienthq/arduino/blob/master/src/AgConfigure.cpp#L60)
  - the operator== can only compare against [other JSONVar](https://github.com/arduino-libraries/Arduino_JSON/blob/master/src/JSONVar.cpp#L328) objects of [null values](https://github.com/arduino-libraries/Arduino_JSON/blob/master/src/JSONVar.cpp#L334)
  - AgConfigure.cpp is trying to [compare the JSONVar to a String](https://github.com/airgradienthq/arduino/blob/master/src/AgConfigure.cpp#L278)
- [src/AgApiClient.cpp:108:7](https://github.com/airgradienthq/arduino/blob/master/src/AgApiClient.cpp#L108): error: 'WiFi' was not declared in this scope
  - WiFi variable is used without being defined in this file nor in AgApiClient.h
[... cut ...] e updated.

From https://www.airgradient.com/blog/install-arduino-c3-mini/ we see:

> Don’t forget to select the Lolin C3 Mini board from Tools > Board > ESP 32 menu after installation.

From comments in the source code we see: Set board to "ESP32C3 Dev Module"

https://github.com/airgradienthq/arduino/blob/8a83e408d3c036b089b63c053a11364b41a26092/examples/OneOpenAir/OneOpenAir.ino#L20-L21

Then, to further complicate matters, the Arduino IDE auto-detects the device as an Adafruit QT Py ESP32-C3

mattkelly, 2024-06-22:
@hestiahacker just confirming I'm seeing the exact same build errors as you. Thank you for writing up great instructions! Definitely extremely confusing trying to get started without them.

hestiahacker, 2024-06-22:
Thanks for your comment, @mattkelly. I was starting to wonder if I was in the twilight zone, where the code works fine for everyone except me!

I look forward to being a regular contributor once I get set up. I'm a big fan of documentation, so whenever I can't figure some thing out, there will be a PR after I get it sorted. 🙂

mattkelly, 2024-06-22:
@hestiahacker FWIW if you need help getting unblocked, here's a minimal patch off of `develop` to make compilation/running work:

1. Lazily remove wifi connection check for now
2. Fix types used for comparison
3. C++ doesn't support designated initializers, so don't use one

```c
diff --git a/src/AgApiClient.cpp b/src/AgApiClient.cpp
index 1f7c83b..375508a 100644
--- a/src/AgApiClient.cpp
+++ b/src/AgApiClient.cpp
@@ -2,10 +2,10 @@
 #include "AgConfigure.h"
 #include "AirGradient.h"
 #include "Libraries/Arduino_JSON/src/Arduino_JSON.h"
+#include <WiFiClient.h>
 #ifdef ESP8266
 #include <ESP8266HTTPClient.h>
 #include <ESP8266WiFi.h>
-#include <WiFiClient.h>
 #else
 #include <HTTPClient.h>
 #endif
@@ -105,9 +105,9 @@ bool AgApiClient::postToServer(String data) {
     return true;
   }
 
-  if (WiFi.isConnected() == false) {
-    return false;
-  }
+  //if (WiFi.isConnected() == false) {
+    //return false;
+  //}
 
   String uri =
       "http://hw.airgradient.com/sensors/airgradient:" + ag->deviceId() +
diff --git a/src/AgConfigure.cpp b/src/AgConfigure.cpp
index 0173edb..e8a6e23 100644
--- a/src/AgConfigure.cpp
+++ b/src/AgConfigure.cpp
@@ -238,7 +238,7 @@ bool Configuration::parse(String data, bool isLocal) {
   String lasCtrl = jconfig[jprop_configurationControl];
   if (isLocal) {
     if (JSON.typeof_(root[jprop_configurationControl]) == "string") {
-      String ctrl = root[jprop_configurationControl];
+      const String ctrl = root[jprop_configurationControl];
       if (ctrl ==
               String(CONFIGURATION_CONTROL_NAME
                          [ConfigurationControl::ConfigurationControlBoth]) ||
@@ -274,7 +274,8 @@ bool Configuration::parse(String data, bool isLocal) {
                     jconfig[jprop_configurationControl]);
     }
 
-    if (jconfig[jprop_configurationControl] ==
+    const String ctrl = jconfig[jprop_configurationControl];
+    if (ctrl ==
         String(CONFIGURATION_CONTROL_NAME
                    [ConfigurationControl::ConfigurationControlCloud])) {
       failedMessage = "Monitor set to accept only configuration from the "
@@ -283,7 +284,8 @@ bool Configuration::parse(String data, bool isLocal) {
       return false;
     }
   } else {
-    if (jconfig[jprop_configurationControl] ==
+    const String ctrl = jconfig[jprop_configurationControl];
+    if (ctrl ==
         String(CONFIGURATION_CONTROL_NAME
                    [ConfigurationControl::ConfigurationControlLocal])) {
       failedMessage = "Cloud configure ignored";
diff --git a/src/MqttClient.cpp b/src/MqttClient.cpp
index fcbefea..96c4803 100644
--- a/src/MqttClient.cpp
+++ b/src/MqttClient.cpp
@@ -23,9 +23,8 @@ bool MqttClient::begin(String uri) {
   logInfo("Init uri: " + uri);
 
   /** config esp_mqtt client */
-  esp_mqtt_client_config_t config = {
-      .uri = this->uri.c_str(),
-  };
+  esp_mqtt_client_config_t config = {};
+  config.broker.address.uri = this->uri.c_str();
 
   /** init client */
   client = esp_mqtt_client_init(&config);
```

mattkelly, 2024-07-03:
Yup thanks for the clarification re: source tree - I've been finding my first dive into the Arduino ecosystem pretty confusing tbh! 

I was also seeing the too big sketch error and honestly, I'm not sure how I got around it. I think I did an extra clean build somehow - it was a while ago now. I suspect that there's some cruft buildup happening from multiple repeated builds, not really sure yet.

stablestud, 2024-07-12:
Interesting that you all used Arduino IDE to compile the project,
I use `PlatformIO` and compilation works fine for the `esp32-c3` (AirGradient ONE) environment:
`pio run -e esp32-c3`
but not for the `esp8266` environment:
`pio run -e esp8266` (compilation errors)
It's used by old models so this was not of my concern.

@pnt325 @airgradienthq 
But I wonder what's the correct way of compiling the project then: Arduino IDE or PlatformIO?

To flash the device I do:
1. Press `Boot SW` while plugging cable in
2. `pio run -e esp32-c3 --target upload`
3. Replug cable to reset device
4. Done

stablestud, 2024-07-12:
Additionally I added the following entries to the `platformio.ini` under the `[env:esp32-c3]` tag

```
board_build.flash_mode = qio
board_build.f_flash = 80000000L
```

hestiahacker, 2024-07-18:
Given that the name of the repo is "arduino", I expected the it was expected to be compiled with the Arduino IDE. I'll be interested to see what @airgradienthq has to say about this.

pnt325, 2024-07-24:
Hi @hestiahacker 

You can find it on Arduino libraries with name `AirGradient Air Quality Sensor`. We are develop the library for Arduino, please follow the instruction on head of file to configuration Arduino before build at [here](https://github.com/airgradienthq/arduino/issues/160#issuecomment-2156223909). We are already test worked before release

hestiahacker, 2024-07-25:
Hello @pnt325 

I feel like there's been a miscommunication here. There are two issues that have not been answered.

1. The code does not compile in the Arduino IDE. This has been verified by multiple people. If @mattkelly's [patch](https://github.com/airgradienthq/arduino/issues/160#issuecomment-2184169078) is applied, it still does not compile because the sketch is too big as described [here](https://github.com/airgradienthq/arduino/issues/160#issuecomment-2201208903).
2. Because the OneOpenAir.ino uses the library instead of the code from the repo, it's difficult for contributors to make changes to the library and test them. In order to make changes, we have to modify files in `~/Arduino/libraries/AirGradient_Air_Quality_Sensor/` as described in [this comment](https://github.com/airgradienthq/arduino/issues/160#issuecomment-2201208903), and then if they work, copy those change over to the checkout of this git repo, submit them without end to end testing, hope that they get accepted, and only after they are accepted and a new version of the library is released can contributors fully test their changes. If there's an error, the entire process starts over. There's also a risk of an error when we're copying the code from the libraries directory to the git checkout which wouldn't exist if we could use the code from the git checkout directly.

This ticket has been open for nearly two months and we are still unable to compile the project, which is a very frustrating experience for someone considering volunteering their time to help maintain the firmware.

airgradienthq, 2024-07-25:
@hestiahacker i am using the Arduino software to compile and flash since many months with the master branch and it works just fine for me. 

It's very late here but I will publish my version number and settings here tomorrow.

airgradienthq, 2024-07-27:
Arduino Editor
[image]

Board Manager:
[image]

Here are my Arduino Settings:
[image]

airgradienthq, 2024-07-27:
@hestiahacker I just re-installed my complete environment and followed [these steps](https://www.airgradient.com/documentation/install-arduino-c3-mini/) and 3.1.4 compiles and flashes.

I am on Mac (Ventura 13.1) and cannot test on Linux. So I'm not sure if there is a problem from that side.

hestiahacker, 2024-07-27:
I set up a new computer with Arduino IDE 2.3.2 (Linux), and it doesn't compile there. There are two issues, compiler errors and a space issue. I've fixed the latter (error on my part), but the former requires code changes.

I'm using v3.1.4 of the AirGradient library.

# Three compilation errors

## WiFi not defined

`WiFi` is not declared in `src/AgApiClient.cpp` because neither WiFi.h nor WiFiClient.h are included in AgApiClient.cpp or AgApiClient.h

I don't understand how this is compiling on your machine.

## Ambiguous overload

`error: ambiguous overload for 'operator=='` in AgConfigure.cpp

These errors is just a matter of casting the jconfig values to a Strings. Maybe this is a difference in compilers since it looks like you're using an old version of the Arduino IDE which was doing an implicit cast.

## Designated initializers

`error: 'esp_mqtt_client_config_t' has no non-static data member named 'uri'`

As said previously: "C++ doesn't support designated initializers, so don't use one". The only thing I can think of is that maybe this is another difference between the versions of Arduino that we are using.

## Details
Here are all the details of the compiler errors when trying to compile the master branch 

<details>
<summary>Compiler errors</summary>

```
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp: In member function 'bool Configuration::parse(String, bool)':
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:290:45: error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  290 |     if (jconfig[jprop_configurationControl] ==
In file included from /home/user/.arduino15/packages/esp32/hardware/esp32/3.0.3/cores/esp32/Arduino.h:192,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Main/PrintLog.h:4,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.h:5,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:1:
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.3/cores/esp32/WString.h:211:8: note: candidate: 'bool String::operator==(const String&) const' (reversed)
  211 |   bool operator==(const String &rhs) const {
      |        ^~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.3/cores/esp32/WString.h:214:8: note: candidate: 'bool String::operator==(const char*) const' (reversed)
  214 |   bool operator==(const char *cstr) const {
      |        ^~~~~~~~
In file included from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/JSON.h:25,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/Arduino_JSON.h:23,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:2:
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/JSONVar.h:85:8: note: candidate: 'bool JSONVar::operator==(const JSONVar&) const'
   85 |   bool operator==(const JSONVar& v) const;
      |        ^~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:298:45: error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  298 |     if (jconfig[jprop_configurationControl] ==
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.3/cores/esp32/WString.h:211:8: note: candidate: 'bool String::operator==(const String&) const' (reversed)
  211 |   bool operator==(const String &rhs) const {
      |        ^~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.3/cores/esp32/WString.h:214:8: note: candidate: 'bool String::operator==(const char*) const' (reversed)
  214 |   bool operator==(const char *cstr) const {
      |        ^~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/JSONVar.h:85:8: note: candidate: 'bool JSONVar::operator==(const JSONVar&) const'
   85 |   bool operator==(const JSONVar& v) const;
      |        ^~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgApiClient.cpp: In member function 'bool AgApiClient::postToServer(String)':
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgApiClient.cpp:108:7: error: 'WiFi' was not declared in this scope
  108 |   if (WiFi.isConnected() == false) {
      |       ^~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgApiClient.cpp:118:3: error: 'WiFiClient' was not declared in this scope
  118 |   WiFiClient wifiClient;
      |   ^~~~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgApiClient.cpp:120:20: error: 'wifiClient' was not declared in this scope
  120 |   if (client.begin(wifiClient, uri.c_str()) == false) {
      |                    ^~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

</details>

# Sketch too large

This was an error on my part. I had the Partition Scheme set to the default instead of the "Minimal SPIFFS..." option. Once I fixed that, I was able to compile my fork (with @mattkelly's patch). To be clear, I could still not compile the master branch of this repo (airgradienthq/arduino) due to the compilation errors described above.

Note: This Partition Scheme setting got reset to the default when I was switching between the "Wemos D1 Mini" board that [the documentation](https://www.airgradient.com/documentation/install-arduino-c3-mini/) says should be selected and the "ESP32C3 Dev Module" that the comments say should be selected.

stablestud, 2024-07-27:
What about using PlatformIO as official way?
Why was the PlatformIO config file added in the first place?

mng2, 2024-10-10:
I just got my unit today and thought I'd share my experience setting up the build. I followed [these instructions](https://www.airgradient.com/documentation/install-arduino-c3-mini/) and was able to successfully compile on Ubuntu 22.04, Arduino IDE version 2.3.3. (Haven't tried flashing, maybe in a few days.)

There were a few differences, if I remember correctly. 
- I didn't use the json for the boards. The Boards Manager already had an entry for esp32 by Espressif.
- I installed AirGradient library `3.1.9`, I assume it goes with the code version in this repo.
- I had to `pip install pyserial`.

hestiahacker, 2024-11-30:
It's been 4 months since my last comment in this thread, so I went back to see if this was silently fixed. I updated the arduino repo (this one), updated Arduino IDE to 2.3.3, updated to v3.1.13 of the AirGradent library (installed via the Arduino IDE's library manager). When I tried to verify the sketch, I got lots of new compilation errors.

So I spun up a fresh Debian 12 VM and tried the instructions @mng2 linked to a couple weeks ago.

```sh
git clone https://github.com/airgradienthq/arduino.git

wget https://downloads.arduino.cc/arduino-ide/arduino-ide_2.3.3_Linux_64bit.AppImage
chmod +x ./arduino-ide_2.3.3_Linux_64bit.AppImage 
./arduino-ide_2.3.3_Linux_64bit.AppImage 
# Install esp32 board by Expressif Systems, v 3.0.7 (I also found that no board manager URL was required)
# Install AirGradent library, v3.1.13
# File -> Open... Select ~/arduino/examples/OneOpenAir/OneOpenAir.ino
# Set board to "ESP32C3 Dev Module"
# Enable "USB CDC On Boot"
# Partition scheme "Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)"
# (All other settings mentioned in OneOpenAir.ino were already correct)
# Click the "Verify" button, observe compilation errors
```

If you run the same commands as above, it will demonstrate that the repo does not compile when following [the documentation](https://www.airgradient.com/documentation/install-arduino-c3-mini/). This is as of commit 566f8a63b4eca99d058d56865545612d656db7c1  The compilation errors are related to headers not being included, or ambiguous operator overloads (full log of errors is below).

I'd love for one of the people who believe this to be incorrect to try exactly what is listed above using Debian 12. This should make sure we're on the same page, and if the problem turns out to somehow be Debian specific, we can go from there.  I'd really love to help maintain the firmware, but it's challenging when it doesn't compile out of the box. Next time I'm at a machine where I can easily spin up an Ubuntu 22.04 VM, I'll try reproducing it over there as well.

<details>
<summary>errors that appear when trying to compile using the commands above</summary>

```text
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp: In member function 'void Configuration::saveConfig()':
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:221:3: error: 'File' was not declared in this scope
  221 |   File file = SPIFFS.open(CONFIG_FILE_NAME, "w", true);
      |   ^~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:222:7: error: 'file' was not declared in this scope
  222 |   if (file && !file.isDirectory()) {
      |       ^~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp: In member function 'void Configuration::loadConfig()':
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:246:3: error: 'File' was not declared in this scope
  246 |   File file = SPIFFS.open(CONFIG_FILE_NAME);
      |   ^~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:247:7: error: 'file' was not declared in this scope
  247 |   if (file && !file.isDirectory()) {
      |       ^~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:256:5: error: 'SPIFFS' was not declared in this scope
  256 |     SPIFFS.format();
      |     ^~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp: In member function 'bool Configuration::begin()':
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:333:8: error: 'SPIFFS' was not declared in this scope
  333 |   if (!SPIFFS.begin(true)) {
      |        ^~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp: In member function 'bool Configuration::parse(String, bool)':
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:420:45: error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  420 |     if (jconfig[jprop_configurationControl] ==
In file included from /home/user/.arduino15/packages/esp32/hardware/esp32/3.0.7/cores/esp32/Arduino.h:192,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Main/PrintLog.h:4,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.h:5,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:1:
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.7/cores/esp32/WString.h:211:8: note: candidate: 'bool String::operator==(const String&) const' (reversed)
  211 |   bool operator==(const String &rhs) const {
      |        ^~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.7/cores/esp32/WString.h:214:8: note: candidate: 'bool String::operator==(const char*) const' (reversed)
  214 |   bool operator==(const char *cstr) const {
      |        ^~~~~~~~
In file included from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/JSON.h:25,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/Arduino_JSON.h:23,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.h:8:
[... cut ...]
      |       ^~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgApiClient.cpp:121:3: error: 'WiFiClient' was not declared in this scope
  121 |   WiFiClient wifiClient;
      |   ^~~~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgApiClient.cpp:124:20: error: 'wifiClient' was not declared in this scope
  124 |   if (client.begin(wifiClient, uri.c_str()) == false) {
      |                    ^~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

</details>

mng2, 2024-12-02:
@hestiahacker The instructions say to use the esp32 board package version `2.0.11`. That may be what you are missing.

ruffson, 2025-01-09:
Did anyone get anywhere with the compilation of this? I ave the same issues as others have reported but was not able to get anywhere. I need to make a change in the library and thought I can just check the repo but it would not compile.

hestiahacker, 2025-01-13:
It is unfortunate that this has been open for half a year and we still have multiple people who can't compile it.

MallocArray, 2025-01-23:
I was able to get this project to compile today and took notes.  If others can test, we can ping the team to add it to the webpage documentation and/or the notes in the files as well.

---
Install Arduino IDE 
arduino-ide_2.3.4_Windows_64bit

Place a copy of "arduino-master" from the Github repo into your `Documents/Arduino/libraries` folder.  You may have to make the 'libraries' subfolder if not present
For Windows users it must be in this folder for the libraries in the code to be detected properly
Mac should be `Documents/Arduino/libraries` and Linux `~/Arduino/libraries`

Open `examples/OneOpenAir/OneOpenAir.ino` with Arduino IDE

Tools>Board>Board Manager and install "esp32 by Espressif Systems" and ensure you select version 2.0.11 (Possibly go up to 2.0.17)
Tools>Board>"ESP32C3 Dev Module" and set other settings mentioned at the top of `examples\OneOpenAir\OneOpenAir.ino`
Make sure you match each one, as missing something like 'Enable "USB CDC On Boot"' can cause compile errors about Serial.h or other related messages

Compile the code which should be successful

MallocArray, 2025-02-12:
I sent a comprehensive list of all of the different documentation sources that are not in agreement with each other to the support e-mail to see if we can get some attention to this issue.

Hopefully that spurs some conversation.

samuelbles07, 2025-02-15:
Hi everyone, we are currently addressing this problem on multiple environment, will update here once its done.

hestiahacker, 2025-02-22:
Thanks @MallocArray it seems like opening support tickets is the way to get traction on issues. I appreciate you helping to get this unstuck.

hestiahacker, 2025-02-27:
First, I can confirm that the instructions do work on Debian 12 (Arduino 2.3.4, esp32 3.2.0-RC1, AirGradient library 3.2.0). 🎉  Strong work.

Second, this project doesn't compile with the latest version of the esp32 board manager. Unless this project is fixed to be compatible with the current release, I think it should be made more apparent that this project can not be compiled with the latest version of "esp32" (by Expressif). I now see that the version is in bold, but I **still** missed it, as I didn't expect to have to use a legacy version of anything. I've included the details of what happens when people try to use the latest version of this package below. A much better solution would be to update the code to work with either the legacy version or the current release. Then there's no need to document these gotchas and no change of people misreading anything.

Third, I think @MallocArray pretty much summed up my other concerns in [their comment here](https://github.com/airgradienthq/arduino/pull/287#discussion_r1973610057). I agree that this documentation is a step forward, and it does address the consumer's needs, however it does not address the needs of potential developers.

Consider the process one would have to go through in order to contribute changes to the project:

1. take a backup of each file in `~/Arduino/libraries` before they modify anything
2. make changes in `~/Arduino/libraries`
3. test these changes
4. for each file they changed, do a diff between the backup and modified code
5. patch that into a checkout of the repo
6. fix any merge conflicts (because the development branch won't match the release)
7. commit, push and submit a merge request

And even after all that, they wouldn't have compiled the exact code that they're submitting upstream.

I'd be happy to try out additional 

# Trying to use the latest versions of libraries

Following the instructions and getting the latest version of esp32 (3.0.0, 3.1.1 or 3.2.0-RC1) and the AirGradient library (3.2.0) still results in compilation errors.

```
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp: In member function 'bool Configuration::parse(String, bool)':
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:529:45: error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  529 |     if (jconfig[jprop_configurationControl] ==
In file included from /home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/Arduino.h:192,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Main/PrintLog.h:4,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.h:5,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:1:
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.h:211:8: note: candidate: 'bool String::operator==(const String&) const' (reversed)
  211 |   bool operator==(const String &rhs) const {
      |        ^~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.h:214:8: note: candidate: 'bool String::operator==(const char*) const' (reversed)
  214 |   bool operator==(const char *cstr) const {
      |        ^~~~~~~~
In file included from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/JSON.h:25,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/Arduino_JSON.h:23,
                 from /home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.h:8:
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/JSONVar.h:85:8: note: candidate: 'bool JSONVar::operator==(const JSONVar&) const'
   85 |   bool operator==(const JSONVar& v) const;
      |        ^~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/AgConfigure.cpp:537:45: error: ambiguous overload for 'operator==' (operand types are 'JSONVar' and 'String')
  537 |     if (jconfig[jprop_configurationControl] ==
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.h:211:8: note: candidate: 'bool String::operator==(const String&) const' (reversed)
  211 |   bool operator==(const String &rhs) const {
      |        ^~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/WString.h:214:8: note: candidate: 'bool String::operator==(const char*) const' (reversed)
  214 |   bool operator==(const char *cstr) const {
      |        ^~~~~~~~
/home/user/Arduino/libraries/AirGradient_Air_Quality_Sensor/src/Libraries/Arduino_JSON/src/JSONVar.h:85:8: note: candidate: 'bool JSONVar::operator==(const JSONVar&) const'
   85 |   bool operator==(const JSONVar& v) const;
      |        ^~~~~~~~

exit status 1

Compilation error: exit status 1
```

It looks like a small change to disambiguate the operator== functions should resolve this and make this project compatible with any version of the esp32 board manager. I'd be willing to try to patch this up one there is a development process which is not as cumbersome as what I laid out above.

MallocArray, 2025-02-27:
Another user in the Forums also had some minor changes that made it able to compile with the 3.x line of the ESP32 board manager:
https://forum.airgradient.com/t/firmware-compile-error/3608/4

When I brought this up earlier, it was mentioned that the AirGradient developers user PlatformIO and it only uses the 2.x line, and I got the impression there was no desire to change to something that would use the 3.x line.

samuelbles07, 2025-02-28:
@hestiahacker 

For your second point, we internally is using PlatformIO for development as @MallocArray said. Unfortunatelly, PlatformIO will not get support for arduino-esp32 v3.x (ESP-IDF > v5.x) as you can see from this [discussion](https://github.com/platformio/platform-espressif32/issues/1225). In my opinion, to support the recent arduino-esp32 version, it's not just about successfully compile the project, because it is a major upgrade change, and i'm afraid there's a runtime problem in the way, so tests needed for this. We have other plan to just use arduino-cli to be able support the recent arduino-esp32 version and not stuck to use arduino ide.

And third point, i already add new section on how to contribute, its a very simple 'howto', but should suffice for now. 

BTW, the docs already on master branch, so others can directly see it without goes to specific branch.

hestiahacker, 2025-02-28:
The instructions to compile now work out of the box. Thank you for adding the extra warning and adding the apt command to install python3-pyserial. I'm closing this ticket as completed. I'm going to submit a merge request for some other changes, but I consider them to be improvements and beyond the scope of this ticket.

Thank you @samuelbles07, @MallocArray and everyone else who contributed to the conversation here. ❤

closed 2025-02-28 by hestiahacker