Open-Air-Foundation/firmware-one-openair #174: Unable to change configurationControl

joostlek, 2024-06-06:
My monitor is set to `local`, but when executing:
```sh
curl --location --request PUT 'http://10.0.0.115/config' \
--header 'Accept: application/json' \
--header 'Content-Type: application/json' \
--data '{
    "configurationControl": "cloud"
}'
```
It will return
```txt
Monitor set to accept only configuration from the cloud. Use property configurationControl to change.
```
Which is strange because I am doing what it's telling me to do

pnt325, 2024-06-06:
@joostlek 
Thank for reporting, that should be success after changed from `local` to `cloud` with that message.

That message indicate you that after change to `cloud` the configuration only change from cloud, all local configuration will be ignore and return failed excepted `configurationControl `

joostlek, 2024-06-06:
But I would expect that after such request I would get a 200 and know the request itself has been successful. I did notice the lights on my device turned on, so I'm assuming it's fetching data from the cloud, so the request itself worked

pnt325, 2024-06-07:
> But I would expect that after such request I would get a 200 and know the request itself has been successful. I did notice the lights on my device turned on, so I'm assuming it's fetching data from the cloud, so the request itself worked

That is the issue. I will update it soon.

joostlek, 2024-06-07:
Ooh check, I didn't interpret your comments as an acknowledgement of the issue but as an explanation that it was expected behaviour. Thanks :)

pnt325, 2024-06-08:
@joostlek 
Please check PR #175.  And open the issue if still problem

Thanks
Phat

closed 2024-06-08 by pnt325

joostlek, 2024-06-08:
Thanks for helping out!