Open-Air-Foundation/firmware-one-openair #178: Low temperature reading from PMS5003T

airgradienthq, 2024-06-13:
There seem to be very low temperature readings sometimes happening from PMS5003T. Model O-1PP

[image]

I think it might make sense to only allow **temperature and humidity readings** within the specs of the sensor:

• Temperature operating range (–40°C to 100°C)
• Relative humidity operating range (0% to 100%)

See: https://www.mouser.com/datasheet/2/1081/ENS210-1984006.pdf

So basically ignoring values outside that band.

joshschaeffer, 2024-06-20:
Not sure if this helps, but here's how Esphome fixed a similar issue with that sensor:
[https://github.com/esphome/esphome/pull/6100/files](https://github.com/esphome/esphome/pull/6100/files)

airgradienthq, 2024-07-01:
Will be fixed as part of #190

closed 2024-07-01 by airgradienthq