Open-Air-Foundation/firmware-one-openair #181: Interest in Github Action for Arduino Compile?

MallocArray, 2024-06-15:
I've been wanting to expand my knowledge of Github Actions and workflows/pipelines.

I see there is an existing Action that does some separate steps to install the arduino-cli and install libraries and compile code.

I found an article that shows a Github Action specific for compiling sketches that can automate some of those manual install steps
https://blog.arduino.cc/2021/04/09/test-your-arduino-projects-with-github-actions/

It also would unlock the ability to collect some reports about changes in memory size and add comments on PRs about those impacts

Would there be interest in switching to this action type instead of the current steps, or are you content with the current flows?  I'm willing to keep plugging away at it, but if there isn't interest, I'll pass on spending the time.

---

I started playing with it in a fork and made some progress.  I haven't gotten the ESP8266 fully working yet due to the board_options not being valid for this type.

The OneOpenAir is getting very close, but I'm getting some compile errors about unused-function and maybe-uninitialized, that may be legit concerns, but they are causing the workflow to be flagged as Failed.  Doing the same compile in Arduino IDE on my workstation does not show these errors and I haven't yet found a way to tell it to treat these as warnings and not failures.

https://github.com/MallocArray/arduino/blob/master/.github/workflows/check2.yml
https://github.com/MallocArray/arduino/actions/runs/9504487354/job/26197252928

```
Compiling sketch: examples/OneOpenAir
  /home/runner/work/arduino/arduino/examples/OneOpenAir/OneOpenAir.ino:853:13: error: 'void failedHandler(String)' defined but not used [-Werror=unused-function]
   static void failedHandler(String msg) {
               ^~~~~~~~~~~~~
  /home/runner/work/arduino/arduino/examples/OneOpenAir/OneOpenAir.ino: In function 'void updatePm()':
  /home/runner/work/arduino/arduino/examples/OneOpenAir/OneOpenAir.ino:1179:51: error: 'hum' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         ag->sgp41.setCompensationTemperatureHumidity(temp, hum);
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
  /home/runner/work/arduino/arduino/examples/OneOpenAir/OneOpenAir.ino:1179:51: error: 'temp' may be used uninitialized in this function [-Werror=maybe-uninitialized]
  cc1plus: some warnings being treated as errors
```

airgradienthq, 2024-07-01:
@MallocArray I would love to have this more automized but we are currently running at capacity with our resources and don't have any time for looking into this. But if you have any important information to share please put them into here. Hopefully we can expand the team at some point and then look into these kind of improvements.

MallocArray, 2025-01-26:
I submitted #279 that would cover this

closed 2025-02-15 by samuelbles07