Open-Air-Foundation/firmware-one-openair #183: Home Assistant Integration - stuck on already_in_progress if initially inaccessible

asheroto, 2024-06-18:
Hello, I came here from here, let me know if it's the wrong spot:
https://forum.airgradient.com/t/native-airgradient-integration-with-homeassistant/2125

I upgraded the firmware on my AirGradient and added it to Home Assistant. Initially my setup said `Unknown error occurred`. Upon further inspection, I found that I had some firewall rules limiting access. I changed those rules, then tried to connect again. This time I got the error message `already_in_progress`.

[image]

So, two things here:

1.) The `Unknown error occurred` may want to be a bit more verbose, or at the very least say "check the Home Assistant logs for details". You could likely integrate a check to confirm connectivity first, then if connectivity fails, tell the user something like, "We couldn't connect to your device. Check network/switch/firewall rules", etc.
2.) I'm guessing `already_in_progress` appeared because it was stuck on the initial connection. The way I resolved this was to `Restart Server` on Home Assistant, then re-add the AirGradient integration. Although this is a feasible workaround, it might be better for the AirGradient integration setup to "soft fail" where it cancels if the `Unknown error occurred` occurs. This way even if somebody simply had their AirGradient disconnected, it wouldn't get stuck on setup. 😊

Here are the logs:

<details><summary>Details</summary>
<p>

```
2024-06-17 23:50:34.819 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/connector.py", line 1025, in _wrap_create_connection
    return await self._loop.create_connection(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1120, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1102, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1005, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 641, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 681, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 111] Connect call failed ('192.168.0.93', 80)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1262, in async_init
    flow, result = await self._async_init(flow_id, handler, context, data)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1292, in _async_init
    result = await self._async_handle_step(flow, flow.init_step, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 517, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/airgradient/config_flow.py", line 50, in async_step_zeroconf
    await self.client.get_current_measures()
  File "/usr/local/lib/python3.12/site-packages/airgradient/airgradient.py", line 80, in get_current_measures
    response = await self._request("measures/current")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/airgradient/airgradient.py", line 60, in _request
    response = await self.session.request(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/client.py", line 581, in _request
    conn = await self._connector.connect(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/connector.py", line 544, in connect
    proto = await self._create_connection(req, traces, timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/connector.py", line 944, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/connector.py", line 1257, in _create_direct_connection
    raise last_exc
  File "/usr/local/lib/python3.12/site-packages/aiohttp/connector.py", line 1226, in _create_direct_connection
    transp, proto = await self._wrap_create_connection(
[... 107 lines cut ...]
    work_item = work_queue.get(block=True)
2024-06-18 00:05:08.536 CRITICAL (SyncWorker_12) [concurrent.futures] Exception in worker
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 82, in _worker
    work_item = work_queue.get_nowait()
                ^^^^^^^^^^^^^^^^^^^^^^^
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 89, in _worker
    work_item = work_queue.get(block=True)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^
SystemExit
```

</p>
</details> 

---

As a separate comment, perhaps the entity ID prefix should be something more descriptive, like "sensor.airgradient_carbon_dioxide", "sensor.airgradient2_carbon_dioxide". That would be easier to find and remember than `sensor.i_9psl_carbon_dioxide`. 😊

Great job guys, this is much easier than it used to be!

joostlek, 2024-06-20:
I'll probably have to catch some exception better to make sure it shows the right message. That's also the reason why you probably see `already_in_progress` since that happens when you try to start a flow that is already ongoing (but the old one did not stop because the exception didnt capture properly).

About the naming, we try to set the device name to something a user can recognize, so if you for example have an indoor and outdoor monitor and add them both, you would have `sensor.airgradient_temperature` and `sensor.airgradient_temperature_2`, which is harder to distinguish. So that's why we chose to put the model name in there so you know which one is which. You can rename a device, and then HA will ask if you want to update all entities as well and then you can easily change `i_9psl` to whatever you like

airgradienthq, 2024-07-01:
@joostlek Please update this ticket and let me know how to proceed with this. Thanks.

joostlek, 2024-09-30:
It's fixed in the latest home assistant release, so this can be closed

asheroto, 2024-10-01:
Will do! Thanks!

closed 2024-10-01 by asheroto