Open-Air-Foundation/firmware-one-openair #188: PCB 3.3 sometimes not recognizing SHT

MallocArray, 2024-06-23:
If this is outside of the scope, it can be closed, but using the firmware for the v3.7 board on a v3.3 board, I immediately saw a message "Sensor init Error: SHT not found" when using the USB-C power port on the board.

When connected to the D1 Mini itself with Micro-USB, it seems to be recognizing it.  So I connected the USB-C power port and one time it initialized correct, but on removing power and connecting it again it failed to initialize.

I've connected multiple times through the Micro-USB and it has worked every time so far, so I don't have any logs from a failed attempt.  As far as I know this SHT module is working correctly, as it was working with ESPHome firmware. 

Using a Rapsberry Pi power adapter with USB-C cable built-in, rated for 5.1v and 3.0 A

airgradienthq, 2024-06-25:
@MallocArray Did you test with SHT30 or SHT40?

airgradienthq, 2024-06-25:
I tested with latest code from: https://github.com/airgradienthq/arduino/tree/feature/add-esp8266-examples

And both SHT30 and SHT40 work on my PCB 3.7 with both USB power ports. 

I cannot reproduce this issue. I believe its something specific to your board or a wobbly SHT module. So I will close this ticket.

closed 2024-06-25 by airgradienthq