Open-Air-Foundation/firmware-one-openair #190: Ignore parameter values out of range

airgradienthq, 2024-06-27:
It seems the PMS5003T seldomly sends extremely low values.

See:
https://github.com/home-assistant/core/issues/120586

I would suggest when reading values from sensor we accept only the following ranges:

Temperature: -40°C to 125°C
Relative Humidity: 0% - 100%
CO2: 0-10.000ppm
PM1, 2.5, 10: 0 - 1000 ug/m3

Byter09, 2024-06-27:
There seems to be a pattern to this though. Every time my temperature goes to `-1000.99`, the PM sensors also go to `-1 µg/m³` or `particles/dL` for the 0.3 PM sensor. The humidity sensor also goes to -1%.

I do not observe this behaviour with the CO2 value, however I'm sure it can't hurt to limit it to sane values as well :)

What I'm trying to say is that maybe there's something that triggers these values to happen at the same time. Let me know if there's anything I can do to help diagnose this.

airgradienthq, 2024-07-01:
@Byter09 Yes that makes sense. The temperature/humidity in the outdoor monitor comes from the PM sensor. So that PM/temperature/humidity fails at the same time indicates a com problem.

Byter09, 2024-07-01:
Here's one more weird quirk I noticed. My internet was off over the night (maintenance by my ISP) until a few minutes ago.

In that whole time, instead of spiking once per day, it spiked several times. I have the feeling this might somehow be related to it not being able to reach the server? Maybe it restarts and the initial value is weird? I don't know. Just leaving this here in the hope that it helps

[image]

mhlas7, 2024-07-11:
Adding another data point here, I turned off "Post Data to AirGradient" two days in Home Assistant (my sensor is set to local control, not cloud) and I now see the temp read -840 F for a two hour period at least once per day. Before I turned it off, I was getting one off spikes down to -840 F similar to other users in this thread. This makes me think the post data to AirGradient function has some sort of impact on the temp reading.

[image]

gslender, 2024-07-22:
Yep, from looking at the data I have, it looks to occur during period of when the sensor is updating to AirGradient in the cloud. Is it possible the internal serial read from the sensor is failing to occur when it should, and bad values are being read instead??

michaelpeterswa, 2024-07-22:
Seeing a similar pattern, posting to AirGradient is enabled, and all data collection for this graph is happening via the local API. 
[image]

If there's anything I can do to assist just let me know.

pnt325, 2024-08-26:
The issue is fixed by #197. Please open issue again if still problem.

closed 2024-08-26 by pnt325

dfsmith, 2024-09-01:
My (new) outdoor Airgradient went funny when I moved it from inside to outside.  I suspect this might be a power issue, and will play with capacitors.
[image]

Byter09, 2024-09-01:
@dfsmith I had something similar happen when the wire wasn't completely attached. Maybe give that a try? Though in my case the sensor reported other values fine, so maybe figure out which of the sensors are behaving weird. If it's all one specific sensor then it's likely the cable.

dfsmith, 2024-09-09:
@Byter09 FYI: 10uF of capacitance on the 3.3V line cleared it right up.  I conjecture that WiFi was stealing too much instantaneous power in the new location, though I didn't collect any hard data to confirm it.

BkSouX, 2024-09-11:
Hi, still had the issue with the air gradient outdoor 

[image]

NorthboundPaddler, 2024-12-03:
Hello - I am still having this issue as well with my O-1PST reporting insane values in Home Assistant. My device is running firmware 3.1.3 and I'll note that the weather here locally is below 0 degrees C, if that matters. If I restart my Home Assistant instance, the values are normal for a few minutes before going well out of range.

Temperature graph is predominantly out of range: 
[image]

Would this be resolved with new firmware? If so, should I use the unified 3.1.9 version or something else?

Thanks for work on this project!

samuelbles07, 2024-12-03:
Hi @NorthboundPaddler , you should use the latest firmware version 3.1.13

NorthboundPaddler, 2024-12-08:
> Hi @NorthboundPaddler , you should use the latest firmware version 3.1.13

I did exactly that today, and it's been working perfectly since I upgraded to this new firmware. Glad that's all it took... I appreciate that you took the time to respond. Cheers!