Open-Air-Foundation/firmware-one-openair #192: Timeout handling

airgradienthq, 2024-07-01:
@pnt325 How does the current firmware handle a time out after a post request to the AirGradient server? Do you show any error message / LED? If so immediately or after a number of fails?

pnt325, 2024-07-08:
@airgradienthq Currently there are no indicate if post failed.

airgradienthq, 2024-07-10:
We have irregular cases where the monitor is showing an error but it recovers very quickly. How often do you retry the request before showing an error?

pnt325, 2024-07-11:
@airgradienthq The sensor S8 and PMS5003 has check number of failed times to set sensor failed(invalid value). The server post failed without any count before show status on display, it's update as the display update period 2500ms.

Could you please details what is the error?

airgradienthq, 2024-08-03:
@pnt325 I do not understand your comment. Please rephrase or use translation software.

pnt325, 2024-08-05:
> We have irregular cases where the monitor is showing an error but it recovers very quickly. How often do you retry the request before showing an error?

S8 and PMS5003/PMS5003T has check number of failed time before response the invalid value storage on [Measurements](https://github.com/airgradienthq/arduino/blob/develop/src/AgValue.h). The value sync to cloud or screen is get from `Measurements`

## S8 sensor 
https://github.com/airgradienthq/arduino/blob/6c1c9147164459bae8e1a4ffd3ea63de6d46ae04/examples/OneOpenAir/OneOpenAir.ino#L318-L331

## PMS5003 sensor

https://github.com/airgradienthq/arduino/blob/6c1c9147164459bae8e1a4ffd3ea63de6d46ae04/examples/OneOpenAir/OneOpenAir.ino#L1007-L1028