Open-Air-Foundation/firmware-one-openair #194: Use HTPPS Endpoints

airgradienthq, 2024-07-08:
Endpoint hw.airgradient.com supports HTTP**S**.

We should switch all endpoints to HTTPS.

pnt325, 2024-08-13:
@airgradienthq 
I have success connect to [hw.airgradient.com](hw.airgradient.com) success on ESP32C3. ESP8266 unable connected to with the same way `self-signed certificate`. I'm run the original example code [BearSSL_Validation.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_Validation/BearSSL_Validation.ino) 

DebugLog
```
13:13:43.572 -> It is also possible to accept *any* self-signed certificate.  This is
13:13:43.572 -> absolutely insecure as anyone can make a self-signed certificate.
13:13:43.572 -> First, try and connect to a badssl.com self-signed website (will fail):
13:13:43.572 -> Trying: self-signed.badssl.com:443...*** Can't connect. ***
13:13:58.588 -> -------
13:13:58.588 -> Now we'll enable self-signed certs (will pass)
13:13:58.634 -> Trying: hw.airgradient.com:443...*** Can't connect. ***
13:13:59.459 -> -------
```

samuelbles07, 2025-02-15:
HTTP applied from 3.1.13 now

closed 2025-02-15 by samuelbles07