Open-Air-Foundation/firmware-one-openair #202: I-9PSL displays dash/hyphen for PM2.5 intermittently when air quality is good

hsugene, 2024-07-21:
Hello, I was referred here from the official AirGradient forum ([here](https://forum.airgradient.com/t/i-9psl-displays-dash-hyphen-for-pm2-5-intermittently-when-air-quality-is-good/2331)). I did a little more digging, so I’ll try to rephrase what I wrote there.

The basic issue is that, when the PM sensor on my I-9PSL is seeing all zero values, there is some flaky behavior:

- The display on the I-9PSL pops up “–“ on the PM2.5 reading for a bit (a few seconds up to about a minute)
- The local monitor API returns no PM readings in the JSON while this is happening

I have been able to reproduce this issue consistently by turning on my air purifier, which is apparently very good at filtering out all the PM in my room. Moreover, when turn off the air purifier and allow the PM levels to rise, the issue does not occur. 

I’m attaching a video file that shows the issue. It’s a 15-minute recording (sped up 20x) showing both the display going from 0 to – and the corresponding local API output from `/measures/current`. I took the liberty of drawing a timeline at the bottom so that it’s easy to identify where the issue occurs. 

To demonstrate normal operation, I opened my windows during the last 2/3 of the video to allow the PM numbers to rise. After doing so, the display and local monitor API provides the expected results reliably.

I can only speculate that, when all PM readings are zero, there’s something that makes the device think “hmm, that can’t be right, so I’m going to reset myself.” But obviously, I’m not the expert here!

Lastly, I’m also attaching the output that is shown in the video.

This isn’t a huge deal, I suppose, since I can just assume that blank values == good air. It just makes the any logs look a bit messy, that’s all. 

Thanks!

https://github.com/user-attachments/assets/778936a9-535d-48ba-9d63-41830868b0f6

[agoutput.txt](https://github.com/user-attachments/files/16325786/agoutput.txt)

pnt325, 2024-07-24:
Hi @hsugene 
If display show `-` mean communication with PM sensor failed. Some time we also get issue. Could you please check the sensor connection to PCB. If so you have two or more PM sensor please attach and do some double check to make sure the sensor is worked well.

From your log it's look the communication to PM sensor always failed.

hsugene, 2024-07-24:
Hi @pnt325, thank you for the reply. I can try to check the connection, but I bought a pre-assembled version and I’m honestly not very comfortable with opening it. I suppose it’s possible that something got knocked loose in transport.

With that said, I do not believe it’s a connection issue because, if I open my windows and let some outside air into the room, the PM sensor will report data reliably the entire time (like 12+ hours). The issue only occurs when the PM levels in the room are close to zero. The way I can reproduce it is by shutting my windows and turning on my air purifier for an hour or so. 

Achim [on the AirGradient forum](https://forum.airgradient.com/t/i-9psl-displays-dash-hyphen-for-pm2-5-intermittently-when-air-quality-is-good/2331) also mentioned that this was not an isolated issue:

> In general this has been an issue we have been fighting with for quite some time. Normally we show a “-” when we get an error from the PM sensor. It seems that at very very low concentrations the sensor module itself sends that error message instead of the correct (very low) reading.

> I think what we have already implemented is a 3x retry in these cases that seem to help a lot but might not catch all cases.

Given this information, do you still think it would be worthwhile to replace the sensor? 

Thank you!

pnt325, 2024-07-25:
Hi @hsugene 
if you have other sensor that would be good to check and compare result. 

And thank for your information we will procedure to test it

airgradienthq, 2024-09-25:
This should be addressed with 3.1.9

closed 2024-09-25 by airgradienthq