Open-Air-Foundation/firmware-one-openair #203: I-9PSL firmware 3.1.4 doesn't recognize SHT3X

pjcreath, 2024-07-25:
At power on, the display says:
```
Sensor init
Error:
SHT not found
```
The sensor works fine in my older board.  I don't know when this regression was introduced.

<details>
<summary>Debug log</summary>

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb 7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x420184b0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
Serial nr: xxxxxxxxxxxx
[Configure] Info: Reading file…
[Configure] Info: Reading file: success
[Configure] Info: Parse file to JSON
[Configure] Info: {“country”:“US”,“pmStandard”:“us-aqi”,“ledBarMode”:“co2”,“abcDays”:8,“tvocLearningOffset”:12,“noxLearningOffset”:12,“mqttBrokerUrl”:"",“temperatureUnit”:“f”,“configurationControl”:“both”,“postDataToAirGradient”:true,“ledBarBrightness”:1,“displayBrightness”:1,“offlineMode”:false,“model”:“I-9PSL”}
Detected ONE_INDOOR
[OledDisplay] Info: begin
Firmware Version: 3.1.4
LED bar brightness: 1
Init SGP41 success
SHTx sensor not found
Set S8 AbcDays successful
[ApiClient] Info: begin
...
```
</details>

pnt325, 2024-08-07:
@pjcreath 
Thank for report the issue. we'll check it.

pnt325, 2024-08-15:
@pjcreath 
I have some code on branch: https://github.com/airgradienthq/arduino/tree/hotfix/sht-sensor-failed to optimize i2c communication that can solve sht3x sensor failed.

pjcreath, 2024-08-15:
@pnt325 
Do you mean commit [e0514e](https://github.com/airgradienthq/arduino/commit/e0514eb0456db317bdbc28101827dcbfd3d3b229)?

Were you able to confirm that this resolved the issue, or do you need me to install the firmware and test?  If the latter, where can I download an image?

Why was SGP originally in a separate task?  Does this change break ESP32 boards?

pnt325, 2024-08-15:
@pjcreath 

> Do you mean commit [e0514e](https://github.com/airgradienthq/arduino/commit/e0514eb0456db317bdbc28101827dcbfd3d3b229)?

Yes

> Why was SGP originally in a separate task? Does this change break ESP32 boards?

It's need to run on runtime as second period to calculate the NOx and VOC

pjcreath, 2024-08-19:
> I still not sure it's solve or not cause I don't have SHT3x sensor to test.

Is there an OTA update with this change?

pjcreath, 2024-10-21:
Still a issue in 3.1.9.