Open-Air-Foundation/firmware-one-openair #207: Update AQI Breakpoints

MallocArray, 2024-08-01:
In the ESPHome project, a user brought up that EPA updated the AQI breakpoints earlier this year.

https://github.com/MallocArray/airgradient_esphome/issues/74

Consider updating formula in AirGradient firmware and/or Dashboard to reflect this change, if not done already

https://www.epa.gov/system/files/documents/2024-02/pm-naaqs-air-quality-index-fact-sheet.pdf
https://en.wikipedia.org/wiki/Air_quality_index#Computing_the_AQI

airgradienthq, 2024-09-02:
We need to synchronize this with an update on our dashboard and also probably change the code for the PM colors on the LEDBar.

MallocArray, 2024-09-02:
Why would the PM colors be adjusted? If you are calculating the compensated values on the device, then what it sent up to the server and the LED lights should be unaffected, as you are now using a more correct value, but still reflecting the same outcome.

lsemprini, 2025-01-15:
apropos...this pull request was merged into development but not sure where it got deployed https://github.com/airgradienthq/arduino/pull/208

closed 2025-01-23 by MallocArray