Open-Air-Foundation/firmware-one-openair #216: Inform AG server of configuration source setting

airgradienthq, 2024-08-24:
If the user set the device up on home assistant and controls it from there, the configuration from airgradient server is not accepted because it is set to "local":

[image]

However, currently there is no notification on the airgradient server on this which confuses users. Can we send the current configuration setting to airgradient server somehow? Maybe as part of the POST measures?

pnt325, 2024-08-25:
Yes we can send the device configuration to cloud like sync measure data to cloud. But from server should accept and store the configuration then if the configuration change form local will be sync to cloud and change from cloud will be send to local

airgradienthq, 2024-09-16:
@nick-4711 Let me know your thoughts on this. Background is that we have a couple of people that set on Home Assistant to ignore cloud configuration but then also use our cloud and try to change things that are then ignored on the device.

So it would be good if we can show a message on our dashboard that the control is local.

What would be the best way to transmit that setting?

nick-4711, 2024-09-16:
I think the device shouldn't attempt to fetch the configuration in the first place, it will discard it anyway. This can then be detected on the server side: The device does post data but never fetches the configuration, this means it is configured through a third party system. It might take some time though until the backend reacts on the configuration change. Maybe this works

- let the device restart if it is switched to local configuration (if not already the case)
- prevent it from posting data until the configuration is fetched

then look at the uptime counter in the data posted to the AG backend, if it is 1 and there was no config fetch in the last few minutes then it is no longer fetching the configuration from the AG backend

airgradienthq, 2024-10-09:
I think this is too complicated and potentially too much delay. Why not we add a parameter or query string to post data about the local configuration?

airgradienthq, 2024-10-09:
Discussed today to go ahead with the server side implementation.

closed 2025-02-15 by samuelbles07