Open-Air-Foundation/firmware-one-openair #227: Data average before sending to the server

GuilhermeZimeo, 2024-08-28:
Hello, Achim explained to me (here: https://forum.airgradient.com/t/export-raw-data-in-original-sampling-frequency/2468/2?u=guilherme) that data is averaged before sending to the server.

However, as far as I could understand the code, such average is only implemented for outdoor monitors with two PM sensors available. For other configurations, it rather seems that raw values are sent to the server every 60 s (without any average) - and the remaining 59 s of data are only used for online logs (Serial.printf)?

Is this intended? Or maybe I have misunderstood something here?
Below are relevant links to the code that led me to this conclusion:

- [targetCount (used for averaging) set to 20 samples 1](https://github.com/airgradienthq/arduino/blob/master/src/AgValue.h#L70)
- [Interval to send data to the server set to 60 s 1](https://github.com/airgradienthq/arduino/blob/master/examples/OneOpenAir/OneOpenAir.ino#L59)
- [Implementation of measurements average 1](https://github.com/airgradienthq/arduino/blob/master/examples/OneOpenAir/OneOpenAir.ino#L1117)
- [Two PMs sensors condition restricting measurements average 1](https://github.com/airgradienthq/arduino/blob/master/examples/OneOpenAir/OneOpenAir.ino#L1117)
- [Not AG One condition also restricting measurements average 1](https://github.com/airgradienthq/arduino/blob/master/examples/OneOpenAir/OneOpenAir.ino#L1032)

airgradienthq, 2024-09-16:
This is definitely something we can implement. I would suggest for version 3.1.9 as we have some more urgent bug fixes to get out of the door before.

samuelbles07, 2025-02-15:
Already solved from version > 3.1.13

We are now using moving average for smoothing with each measurement has a different interval

closed 2025-02-15 by samuelbles07