Open-Air-Foundation/firmware-one-openair #250: Display no longer turns off at night

yesezra, 2024-09-30:
Hello, AirGradient team! I noticed that recently, my AirGradient device no longer dims its display or LEDs at night. I verified that no changes were made in the dashboard. I verified that the firmware is 3.1.9. Please let me know what other information I can send that might help identify this bug. I did recently add my device to a Home Assistant installation - could that be affecting this?

airgradienthq, 2024-10-09:
I cannot reproduce this. Please make sure you did not set your local configuration to only accept local configuration. 

See
https://github.com/airgradienthq/arduino/blob/master/docs/local-server.md

yesezra, 2024-10-09:
Thanks so much for responding. I updated "configurationControl" to "both" and the display is now dimming again. I suspect the Home Assistant integration changed the setting without my knowledge. Closing the issue. Many thanks!

closed 2024-10-09 by yesezra