Open-Air-Foundation/firmware-one-openair #253: BootCount not updated when posting to AirGradient is disabled

dtrejod, 2024-10-06:
## Problem

When using the[ local server](https://github.com/airgradienthq/arduino/blob/master/docs/local-server.md#get-current-air-quality-get) API to get air gradient sensor data, I noticed the `boot` and `bootCount` values were not being updated. This only happened after I configured `postDataToAirGradient: false`. I have a simple airgradient [prometheus exporter](https://github.com/dtrejod/airgradient-exporter) that I used to gather the local server metrics into a Prometheus instance. You can see from the below Grafana graph before `postDataToAirGradient: false`, the boot count metric was incremented every 1m. After the change, the `bootCount` value doesn't increment. 

[image] 

## Analysis

It appears `bootCount` is incremented in the [`sendDataToServer()`](https://github.com/airgradienthq/arduino/blob/c841476ca4d47c8a0005e231a94c8199961f5624/examples/BASIC/BASIC.ino#L551) func. The func returns early, not incrementing `bootCount`, when posting to airgradient servers is disabled.

airgradienthq, 2024-10-09:
Yes good point. We will fix in one of the next updates.

dtrejod, 2025-01-21:
I can confirm #255 fixed the issue. Thanks!

closed 2025-01-21 by dtrejod