Open-Air-Foundation/firmware-one-openair #262: Add combo LED modus

domoritz, 2024-11-10:
I use an air gradient one for indoor air quality tracking. I like he LEDs as a quick status indicator for how things are. https://github.com/MallocArray/airgradient_esphome has a really cool/useful mode where the LEDs together indicate CO2, PM2.5 and VOC. The mode, called "combo" is defined in https://github.com/MallocArray/airgradient_esphome/blob/main/packages/led_combo.yaml.

```
# Combo LED bar
# Left 5 LEDs indicate CO2
# Middle 5 LEDs indicate PM2
# Far right LED indicates TVOC
# LED Fade setting fades out the bars on the sides at the set percentage each step
```

I really like this mode as it makes the LEDs more useful. Could you add a combo mode to the standard firmware?

MallocArray, 2025-06-19:
I'm in favor of this!

27068, 2025-09-25:
Just leaving this here until I find some time to make a PR (also I changed the threshold values which makes it incompatible with upstream).

https://github.com/airgradienthq/arduino/compare/master...27068:airgradienthq-arduino:feat/led