Open-Air-Foundation/firmware-one-openair #265: unsigned short used for temperature causes temps below 0 overflow

blackmagic0, 2024-11-25:
unsinged short is being used to store the calculated the temperature from the sensor instead of a signed short, causing the temperature to report 6553 when the real temperature is below 0

https://github.com/airgradienthq/arduino/blob/79fbd901bd408062e2776c2380105a3176de8053/src/Sgp41/Sgp41.cpp#L316C63-L317C16

samuelbles07, 2024-11-26:
Hi @blackmagic0 , after looking through, its not a problem on SGP41 class, that function have a different purpose. You could update your monitor firmware version to our latest version 3.1.12 and see if the problem still occur. Because there's a restructuring on measurements type.

blackmagic0, 2024-11-29:
> Hi @blackmagic0 , after looking through, its not a problem on SGP41 class, that function have a different purpose. You could update your monitor firmware version to our latest version 3.1.12 and see if the problem still occur. Because there's a restructuring on every measurements type.

Sure, ill give it a go and let you know if it is resolved

blackmagic0, 2024-11-29:
> Hi @blackmagic0 , after looking through, its not a problem on SGP41 class, that function have a different purpose. You could update your monitor firmware version to our latest version 3.1.12 and see if the problem still occur. Because there's a restructuring on every measurements type.

unfortunately I am having trouble updating the firmware by building from source in the Arduino IDE, I followed the instructions [here](https://www.airgradient.com/documentation/open-air-pst-kit-1-3/#arduino), however I am getting the following build error:
```
c:\Users\blackmagic\Documents\Arduino\libraries\AirGradient_Air_Quality_Sensor\src\AgApiClient.cpp: In member function 'bool AgApiClient::postToServer(String)':
c:\Users\blackmagic\Documents\Arduino\libraries\AirGradient_Air_Quality_Sensor\src\AgApiClient.cpp:113:7: error: 'WiFi' was not declared in this scope
  113 |   if (WiFi.isConnected() == false) {
      |       ^~~~
c:\Users\blackmagic\Documents\Arduino\libraries\AirGradient_Air_Quality_Sensor\src\AgApiClient.cpp:121:3: error: 'WiFiClient' was not declared in this scope
  121 |   WiFiClient wifiClient;
      |   ^~~~~~~~~~
c:\Users\blackmagic\Documents\Arduino\libraries\AirGradient_Air_Quality_Sensor\src\AgApiClient.cpp:124:20: error: 'wifiClient' was not declared in this scope
  124 |   if (client.begin(wifiClient, uri.c_str()) == false) {
      |                    ^~~~~~~~~~
```

This is using version 3.1.12 of the Arduino library.
"https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json" added to additional board manager urls
- Board to "ESP32C3 Dev Module"
- Enable "USB CDC On Boot"
- Flash frequency "80Mhz"
- Flash mode "QIO"
- Flash size "4MB"
- Partition scheme "Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)"
- JTAG adapter "Disabled"

I did also try with the board set as the "lolin C3 mini", as per the instructions on the webpage, but the instructions in OneOpenAir.ino say use the other board.

samuelbles07, 2024-11-29:
Just want to make sure, your monitor is open air (outdoor) version right? Why don't you just use our OTA update feature? You can open the airgradient dashboard and go to settings then firmware. There, you can set the target version.
Also, you can see airgradient forum if you want https://forum.airgradient.com/

[image]

blackmagic0, 2024-11-29:
yes, I have the O-1PST.  
Id like to avoid the use of in-browser flash because I don't trust chrome and I don't like the idea of my browser accessing serial devices.  
Id also like to avoid using the dashboard because I don't like the idea of phone-home/telemetry, and I have limited bandwidth in my location.
Ill check the forum and see if I can figure it out.  thanks.

blackmagic0, 2024-11-29:
I was able to build in vscode on linux using platformio to explicitly build for the esp32c3 and upload, the instructions for arduino IDE don't seem to work because it wants to build esp8266 and the error lies in there somewhere.  I was able to confirm that the temperature issue is fixed in firmware version 3.1.13

closed 2024-11-29 by blackmagic0