Open-Air-Foundation/firmware-one-openair #271: Local server host name not following RFC 1034 preferred name syntax

TrodelHD, 2024-12-07:
## Description

I ran into a problem trying to implement the [/docs/local-server.md](https://github.com/airgradienthq/arduino/blob/master/docs/local-server.md) API with the Java [HttpClient](https://docs.oracle.com/en/java/javase/17/docs/api/java.net.http/java/net/http/HttpClient.html). The Problem is that the client straight up refuses to work with a host name that is not compliant with [RFC 1123 - 2.1 Host Names and Numbers](https://www.rfc-editor.org/rfc/rfc1123#section-2) which references [RFC 952 - ASSUMPTIONS](https://www.rfc-editor.org/rfc/rfc952). These RFCs essentially require a host name to only consist of alphabet (a-zA-Z), digits (0-9), minus sign (-), and period (.).

Note that is is a limitation of the Java [HttpClient](https://docs.oracle.com/en/java/javase/17/docs/api/java.net.http/java/net/http/HttpClient.html).

The [RFC 6762](https://datatracker.ietf.org/doc/html/rfc6762) (mDNS RFC) states that it heavily borrows from the existing DNS protocol [[RFC1034](https://datatracker.ietf.org/doc/html/rfc1034)] [[RFC1035](https://datatracker.ietf.org/doc/html/rfc1035)] [[RFC6195](https://datatracker.ietf.org/doc/html/rfc6195)], using the existing DNS message structure, name syntax, and resource record types. [RFC 1034 - 3.5 - Preferred name syntax](https://datatracker.ietf.org/doc/html/rfc1034#section-3.5)  states that you can use any binary string when naming an object in DNS but to avoid problem with many applications it recommends that it should only consist of alphabet (a-zA-Z), digits (0-9), minus sign (-), and period (.).
An example of such a problem would be the Java [HttpClient](https://docs.oracle.com/en/java/javase/17/docs/api/java.net.http/java/net/http/HttpClient.html) refusing to work with `_` in the host name.

## The problem
The local server mDNS host name of an AirGradient Monitor `http://airgradient_{{serialnumber}}.local` contains a `_`. This technically does not violate any RFC standard but results in errors with some other applications (like Java [HttpClient](https://docs.oracle.com/en/java/javase/17/docs/api/java.net.http/java/net/http/HttpClient.html)).

## Potential solution
A simple solution would be to replace the `_` with a `-`. The problem is that this will result in existing clients breaking.
I see three courses of action:

1. Simply replace the  `_` with a `-` and accept that existing clients will break
2. Do nothing because it's not following a preferred (not mandatory) name syntax
3. Add a config that defaults to `_` for existing devices and to for newly setup `-`

# Relevant parts

## Relevant part of [RFC 1034](https://datatracker.ietf.org/doc/html/rfc1034#section-3.5)
```
[3.5]. Preferred name syntax

      The DNS specifications attempt to be as general as possible in the rules
      for constructing domain names.
      ...
      The following syntax will result in fewer problems with many
      applications that use domain names (e.g., mail, TELNET).
      ...
      <domain> ::= <subdomain> | " "
      <subdomain> ::= <label> | <subdomain> "." <label>
      <label> ::= <letter> [ [ <ldh-str> ] <let-dig> ]
      <ldh-str> ::= <let-dig-hyp> | <let-dig-hyp> <ldh-str>
      <let-dig-hyp> ::= <let-dig> | "-"
      <let-dig> ::= <letter> | <digit>
      <letter> ::= any one of the 52 alphabetic characters A through Z in
      upper case and a through z in lower case
      <digit> ::= any one of the ten digits 0 through 9
      ...
      The labels must follow the rules for ARPANET host names.  They must
      start with a letter, end with a letter or digit, and have as interior
      characters only letters, digits, and hyphen.  There are also some
      restrictions on the length.  Labels must be 63 characters or less.
```

## Relevant part of [RFC 1123](https://www.rfc-editor.org/rfc/rfc1123#section-2)
```
5.1  Host Names and Numbers

      The syntax of a legal Internet host name was specified in RFC-952
      [DNS:4].  One aspect of host name syntax is hereby changed: the
      restriction on the first character is relaxed to allow either a
      letter or a digit.  Host software MUST support this more liberal
      syntax.
      ...
```

## Relevant part of [RFC 952](https://www.rfc-editor.org/rfc/rfc952)
```
ASSUMPTIONS

   1. A "name" (Net, Host, Gateway, or Domain name) is a text string up
   to 24 characters drawn from the alphabet (A-Z), digits (0-9), minus
   sign (-), and period (.).  Note that periods are only allowed when
   they serve to delimit components of "domain style names".
   ...
```

samuelbles07, 2024-12-07:
Hi @TrodelHD , there's an easier way i guess. On your router, you can set static IP for the device, based on your device mac address. Then, your java app can directly using IP address.

TrodelHD, 2024-12-08:
@samuelbles07 I’ve already solved this for myself by switching to the [OkHttp](https://mvnrepository.com/artifact/com.squareup.okhttp3/okhttp) client, so this isn’t about finding a solution for me. My goal is to prevent others from wasting hours debugging, only to realize that their programming language’s built-in HttpClient enforces strict rules regarding the host name.

I’d completely understand if addressing this issue feels like too much effort (especially because no rule of [RFC 6762](https://datatracker.ietf.org/doc/html/rfc6762) is violated). My goal was just to bring it to your attention.

A simple solution might be adding a “Known Issues” section to the `docs/local-server.md` file, mentioning that the `_` in the host name can cause problems with application enforcing strict rules on it (e.g., the Java [HttpClient](https://docs.oracle.com/en/java/javase/17/docs/api/java.net.http/java/net/http/HttpClient.html)) and is unavoidable due to legacy reasons.

closed 2024-12-08 by samuelbles07