Open-Air-Foundation/firmware-one-openair #273: Use publicly signed certificate on https://hw.airgradient.com

MallocArray, 2024-12-14:
On the main website, a LetsEncrypt cert is used to be publicly trusted.

On the API endpoint, it is only a self-signed certificate and not one that can be fully trusted by an end client.

Update the cert to also use LetsEncrypt so everything is secure and publicly signed

[image]

[image]

samuelbles07, 2024-12-17:
The `hw` subdomain only used by airgradient monitor (not browser, etc). The problem of using Authorized CA is we need to re-new the certificate every X time, and it will have a problem on the device itself if we don't have a rolling certificate scenario.
As far as i know, as long as the client only accept the certificate from `hw.airgradient.com` domain, i don't think it will have a problem.

MallocArray, 2024-12-17:
If the device trusts the root certificate of Lets Encrypt "ISRG Root X1" and the intermediate CAs, then it will trust any rolling certificate that is issued.  And if you aren't validating the current self signed certificate, then there would be no further harm by using a proper certificate.

It sounds like you don't think security is communication is important, but that feels like just avoiding doing the proper thing.  If traffic to and from the device isn't encrypted and verified to be the proper endpoint, it would be possible for someone to setup a malicious configuration server to send invalid configs to devices, changing calibrations, enabling/disabling LEDs that the user did not intend, etc.

This is the end user's data and it should be protected with reasonable efforts.  You already use Lets Encrypt for the main site, so due diligence should be done to properly protect all of the data.

samuelbles07, 2024-12-17:
> And if you aren't validating the current self signed certificate, then there would be no further harm by using a proper certificate.

I totally agree with you on above part, on the latest firmware on master branch, the certificate is indeed verified. What i mean is, i don't think there any difference when we are using for example Lets Encrypt and self signed certificate when both certificate is validated/verified on the client side.

MallocArray, 2024-12-17:
I run mine with ESPHome firmware and have not been including the self-signed certificate, so validation is not possible.  If it were publicly signed, then ESP-IDF can be used to include root CA bundle from Mozilla which will trust the Lets Encrypt certs.

samuelbles07, 2024-12-17:
I only know little about ESPHome, but if it can include ESP-IDF code on it, then maybe you can include [this root CA](https://github.com/airgradienthq/arduino/blob/master/src/AirGradient.h#L23) for validation.

closed 2025-02-15 by samuelbles07