Open-Air-Foundation/firmware-one-openair #276: Consider moving from SPIFFS to LittleFS

MallocArray, 2025-01-23:
Consider changing from using SPIFFS to LittleFS for these devices.

Currently https://github.com/airgradienthq/arduino/blob/92e74feabdb7518b055df85c74fc694aaad9a45f/src/AgConfigure.cpp#L4 uses SPIFFS.h which ultimately limits us to the ESP board manager 2.x line.  When using Board Manager for ESP v3.x there are multiple errors that stem from SPIFFS not working.

After looking around, it appears that SPIFFS is generally considered depreciated due to not being maintained, and that LittleFS is considered a near drop-in replacement and actively supported:

https://github.com/esp8266/Arduino/issues/7095
https://www.reddit.com/r/esp32/comments/y28m78/spiffs_deprecated_for_littlefs_in_the_past_but_no/

Compiles correctly with ESP32 board v 2.0.11 or 2.0.17
Example of errors with v3.1.1

```
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgConfigure.cpp: In member function 'void Configuration::saveConfig()':
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgConfigure.cpp:221:3: error: 'File' was not declared in this scope
  221 |   File file = SPIFFS.open(CONFIG_FILE_NAME, "w", true);
      |   ^~~~
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgConfigure.cpp:222:7: error: 'file' was not declared in this scope
  222 |   if (file && !file.isDirectory()) {
      |       ^~~~
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgConfigure.cpp: In member function 'void Configuration::loadConfig()':
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgConfigure.cpp:246:3: error: 'File' was not declared in this scope
  246 |   File file = SPIFFS.open(CONFIG_FILE_NAME);
      |   ^~~~
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgConfigure.cpp:247:7: error: 'file' was not declared in this scope
  247 |   if (file && !file.isDirectory()) {
      |       ^~~~
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgConfigure.cpp:256:5: error: 'SPIFFS' was not declared in this scope
  256 |     SPIFFS.format();
      |     ^~~~~~
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgConfigure.cpp: In member function 'bool Configuration::begin()':
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgConfigure.cpp:333:8: error: 'SPIFFS' was not declared in this scope
  333 |   if (!SPIFFS.begin(true)) {
      |        ^~~~~~
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgApiClient.cpp: In member function 'bool AgApiClient::postToServer(String)':
C:\Users\WDAGUtilityAccount\Documents\Arduino\libraries\arduino-master\src\AgApiClient.cpp:122:7: error: 'WiFi' was not declared in this scope
  122 |   if (WiFi.isConnected() == false) {
      |       ^~~~

exit status 1

Compilation error: exit status 1
```

samuelbles07, 2025-02-15:
Hi @MallocArray ,

Thanks for pointing this out! 

Our development is using PlatformIO and latest arduino-esp32 version that PlatformIO use is 2.0.17. And from discussion [here](https://github.com/platformio/platform-espressif32/issues/1225) looks like PlatformIO will not get the latest arduino-esp32 version (in this case 3.x).

So i don't think there's a real benefit to move to LittleFS (for now). Unless we decide to use other than PlatformIO or there's an update about this from espressif side.

MallocArray, 2025-02-15:
ok.  I'm not familiar with what PlatformIO brings to the table, just concerned that if you are staying on 2.x that isn't regularly updated, it will fall behind.

When other users are trying to compile firmware on their own, it is a manual choice to downgrade to 2.x since the latest 3.x is automatically selected in Arduino IDE

Thank you for at least reading it and considering.  You have your reasons to stay where you are and with that I'll close this.

closed 2025-02-15 by MallocArray