Open-Air-Foundation/firmware-one-openair #284: Firmware version confusion

AlexanderLeporiday, 2025-02-06:
Releases show 3.1.13, website shows 3.1.13.
Dashboard shows 3.1.21 and just updated brand new (previously locally connected) I-9PSL.

I had it connected to home assistant and showed it shipped with 3.1.11, manually updated to 3.1.13 from the website.
Connected it to home assistant and shows 3.1.21 was ready.
Connected to cloud, restarted, automatically updated to 3.1.21.
Reset to local mode and home assistant says it's working as expected.

So why have none of the versions since 3.1.13 been marked as releases?

samuelbles07, 2025-02-07:
There's a problem on our versioning now that needs to be more managed properly in the future. We try to improve it little by little until release pipeline better in the future.

MallocArray, 2025-02-12:
To make matters worse, now on the Dashboard, it appears that 3.1.21 was pulled, as the most recent shown in 3.1.13

[image]

But the Releases here in Github now shows 3.2.0

AlexanderLeporiday, 2025-02-12:
3.2.0 was pushed to release 3 days ago but so far hasn't shown up anywhere.
The API (which I can't find right now) where you put in your device serial number and it returns the latest version seems so still return 3.1.21.
[image]

Website still shows Version 3.1.13 Release which was the latest version pushed to GitHub prior to 3.2.0.
https://www.airgradient.com/documentation/firmwares/

Release pipelines seems to be fairly fundamental and you probably shouldn't be maintaining and supporting multiple pathways to release firmware anyway. Make a CI/CD pipeline and then hook in a release system, however you want to do it but pick a lane and stick to it.

washley, 2025-04-25:
My new device tries to OTA update to 3.2.0 on every boot and fails most of the time. So even if I manually update it to 3.3.6 (the latest), eventually it manages to successfully downgrade itself to 3.2.0, after which it doesn't post data until I re-upgrade it. This is super frustrating.

MallocArray, 2025-04-25:
If you are connected to the Dashboard, it has a setting to auto upgrade you to the version that they define as the latest, or you can pick a particular version. 

If 3.2.0 is the version pushed on the dashboard, then when you upgrade manually, it sees you aren't running what the dashboard says, so it brings you back. 
Check about disabling the upgrade in the dashboard, or if you are compiling it yourself you can add "-snap" to the firmware number to keep it from auto upgrading

washley, 2025-04-25:
Unchecking "Always update to the newest version" had no effect, and the version selection is a drop-down that maxes out at 3.1.13 and has no option for me to type anything in.

I'm waiting for post approval to the support forum because there's also the issue of OTA updates almost always failing, too.