Open-Air-Foundation/firmware-one-openair #285: Config value "disableCloudConnection" is not set

lschloetterer, 2025-02-16:
I'm trying set the config like this:
`curl -X PUT -H "Content-Type: application/json" -d '{"disableCloudConnection": true}' http://192.168.100.137/config`
`Success`

However, the value is not changed:
`curl http://192.168.100.137/config | jq .disableCloudConnection`
`false`

It seems to me that somewhere in `AgConfigure.cpp` the variable assignment is missing

samuelbles07, 2025-02-17:
`disableCloudConnection` is a read only configuration. Please see [here](https://github.com/airgradienthq/arduino/blob/master/docs/local-server.md#configuration-parameters-getput) under notes after the table.

But thank you for pointing this, response is misleading

lschloetterer, 2025-02-17:
I see, thank you. I must have only read the PR description instead of the docs.
Sorry about that.

After a factory reset the variable can be set on the WiFi config page without a problem.

closed 2025-02-17 by lschloetterer