Open-Air-Foundation/firmware-one-openair #307: symlinks to other Library repos breaking the ability to clone recursively

MallocArray, 2025-04-28:
In the page for how to compile the firmware, it gives instructions for how to clone the repository locally for the Arduino Library for local development

https://github.com/airgradienthq/arduino/blob/master/docs/howto-compile.md

It has the `--recursive` flag on the git clone command, but currently, this results in errors

```
C:\Users\WDAGUtilityAccount\Documents>git clone --recursive https://github.com/airgradienthq/arduino.git AirGradient_Air_Quality_Sensor
Cloning into 'AirGradient_Air_Quality_Sensor'...
remote: Enumerating objects: 6145, done.
remote: Counting objects: 100% (2038/2038), done.
remote: Compressing objects: 100% (593/593), done.
remote: Total 6145 (delta 1727), reused 1476 (delta 1445), pack-reused 4107 (from 3)
Receiving objects: 100% (6145/6145), 16.01 MiB | 28.36 MiB/s, done.
Resolving deltas: 100% (3901/3901), done.
Submodule 'src/Libraries/airgradient-client' (git@github.com:airgradienthq/airgradient-client.git) registered for path 'src/Libraries/airgradient-client'
Submodule 'src/Libraries/airgradient-ota' (git@github.com:airgradienthq/airgradient-ota.git) registered for path 'src/Libraries/airgradient-ota'
Cloning into 'C:/Users/WDAGUtilityAccount/Documents/AirGradient_Air_Quality_Sensor/src/Libraries/airgradient-client'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:airgradienthq/airgradient-client.git' into submodule path 'C:/Users/WDAGUtilityAccount/Documents/AirGradient_Air_Quality_Sensor/src/Libraries/airgradient-client' failed
Failed to clone 'src/Libraries/airgradient-client'. Retry scheduled
Cloning into 'C:/Users/WDAGUtilityAccount/Documents/AirGradient_Air_Quality_Sensor/src/Libraries/airgradient-ota'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:airgradienthq/airgradient-ota.git' into submodule path 'C:/Users/WDAGUtilityAccount/Documents/AirGradient_Air_Quality_Sensor/src/Libraries/airgradient-ota' failed
Failed to clone 'src/Libraries/airgradient-ota'. Retry scheduled
Cloning into 'C:/Users/WDAGUtilityAccount/Documents/AirGradient_Air_Quality_Sensor/src/Libraries/airgradient-client'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:airgradienthq/airgradient-client.git' into submodule path 'C:/Users/WDAGUtilityAccount/Documents/AirGradient_Air_Quality_Sensor/src/Libraries/airgradient-client' failed
Failed to clone 'src/Libraries/airgradient-client' a second time, aborting
```

samuelbles07, 2025-05-01:
Hi @MallocArray , hmm its interesting that it say doesn't have correct access. Could you try to clone just for test one of the submodule directly?

`git clone https://github.com/airgradienthq/airgradient-client`

ccoley, 2025-05-01:
@samuelbles07 
The issue here is probably that @MallocArray is cloning the repo on a machine that doesn't have a SSH key registered in their Github account. The initial clone works because it's happening over HTTPS, but the submodules are defined with SSH remotes in the `.gitmodules` file so those clones attempt to happen over SSH.

@MallocArray 
Can you test my fix in https://github.com/airgradienthq/arduino/pull/311

Instructions on how to test are present in the pull request description.

MallocArray, 2025-05-01:
@ccoley that sounds reasonable to me, as I don't use SSH for any github stuff and wouldn't have a key with access to the AirGradient repos.

I ran the test of the HTTPS clone from your branch successfully.

As I don't use SSH, I don't have a key or experience to with it.

MallocArray, 2025-05-01:
> Hi [@MallocArray](https://github.com/MallocArray) , hmm its interesting that it say doesn't have correct access while the repo is actually public. Could you try to clone just for test one of the submodule directly?
> 
> `git clone git@github.com:airgradienthq/airgradient-client.git`

This test fails, but as ccoley mentioned, I'm not using SSH or have any keys setup for git.  For my testing I'm using a blank new Windows 11 VM with just the Git for Windows client installed and then attempting to clone.

```
git clone git@github.com:airgradienthq/airgradient-client.git
Cloning into 'airgradient-client'...
The authenticity of host 'github.com (140.82.114.4)' can't be established.
ED25519 key fingerprint is SHA256:+DiY3wvvV6TuJJhbpZisF/zLDA0zPMSvHdkr4UvCOqU.
This key is not known by any other names.
Are you sure you want to continue connecting (yes/no/[fingerprint])? y
Please type 'yes', 'no' or the fingerprint: yes
Warning: Permanently added 'github.com' (ED25519) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

samuelbles07, 2025-05-04:
Thank you @MallocArray and @ccoley . #311 is merged.

PR #311 (2025-05-04): Fix failing to clone submodules recursively
This should fix #307 by using a relative path for the submodule URLs. Users that clone the repo with HTTPS will also clone the submodules with HTTPS, and users that clone the repo with SSH will clone the submodules with SSH.

We keep the username in the path so that forks don't need to also fork the submodules.

For example, the configuration below would not work for a user that forks only the main repo and not the submodules because git would look for the submodules in the namespace of the forking user.

```
[submodule "src/Libraries/airgradient-cli [... cut]
closed 2025-05-04 by samuelbles07