Open-Air-Foundation/firmware-one-openair #308: UTE update to release 3.3.6 broke connectivity to WiFi

PatrickAtBritpol, 2025-04-29:
After recent upgrade my AirGradient ONE (I-9PSL) stopped connecting to WiFi (2.4 & 5 Ghz with the same SSID available). I have tried to configure WiFi settings multiple times. That included clearing all the settings to default values too.

When the device is in offline mode it displays the measurements correctly without a problem. When in online mode keeps on rebooting (I'm not sure if it makes any difference but locally I have my own Grafana and Prometheus querying the device) and it was configured NOT to send data to AirGradient servers.

I have attempted to install the software over USB following https://www.airgradient.com/documentation/firmwares/ (just in case the OTE had some issue) . The install went with no problems. I have rebooted the device.

The screens followed:
1. AirGradient ONE FW Version: 3.3.6
2. Press now for LED test
3. Monitor initialising...
4. Press now for offline mode
5. Initialise network...
6. Connecting to WiFi...
7. Warming Up Serial numer: _actual_serial_number_
8. The proper values columns screen with dashes for numbers (lasts for about 3 sec.)
9. AirGradient ONE FW Version: 3.3.6
10. Press now for LED test
11. Monitor initialising...
12. Press now for offline mode
13. Initialise network...
14. Connecting to WiFi...
15. 180s to connect to WiFi hotspot: "airgradient-_actual_serial_number_"
16. ->reboot

After long poweroff the sequence of screens goes:
1. AirGradient ONE FW Version: 3.3.6
2. Press now for LED test
3. Monitor initialising...
4. Press now for offline mode
5. Initialise network...
6. Connecting to WiFi...
7. Firmware Update up to date
8. AirGradient ONE FW Version: 3.3.6
9. Press now for LED test
10. Monitor initialising...
11. Press now for offline mode
12. Initialise network...
13. Connecting to WiFi...
14. 180s to connect to WiFi hotspot: "airgradient-_actual_serial_number_"
15. ->reboot

I am able to connect to hotspot and interact with the settings screen.

ccoley, 2025-04-30:
This seems to be related to this issue: https://github.com/airgradienthq/arduino/issues/309

The devices seem to reset when you make a request to the `/metrics` endpoint, and Prometheus is polling that endpoint. For now you can stop Prometheus from polling the devices, or downgrade to v3.2.0.

Note that setting the firmware to v3.2.0 in the Airgradient dashboard won't work unless your Prometheus server is polling slow enough to allow the device to connect to Airgradient's servers, receive the command to downgrade the firmware, and then complete the downgrade. So either remove the devices from Prometheus temporarily, or manually flash the older firmware.

samuelbles07, 2025-05-05:
Hi @PatrickAtBritpol , could you confirm this?

danielmoore-sdx, 2025-05-05:
I had the same problem, and it's been fixed by flashing v3.2.0

PatrickAtBritpol, 2025-08-25:
Hi @samuelbles07 , after summer holiday I got back to the project. The device upgraded OTE to 3.3.9 - no problems, all back to normal.