Open-Air-Foundation/firmware-one-openair #309: I-9PSL unit reset itself when calling /metrics endpoint

m4r1k, 2025-04-30:
Hey there,

When curling the `/metrics` endpoint, much to my surprise, the AirGradient ONE reset itself.

AirGradient ONE DIY (I-9PSL)
Firmware: 3.3.6

config
```json
{
  "country": "IT",
  "pmStandard": "ugm3",
  "ledBarMode": "co2",
  "abcDays": 8,
  "tvocLearningOffset": 12,
  "noxLearningOffset": 12,
  "mqttBrokerUrl": "",
  "temperatureUnit": "c",
  "configurationControl": "local",
  "postDataToAirGradient": true,
  "ledBarBrightness": 100,
  "displayBrightness": 100,
  "offlineMode": false,
  "monitorDisplayCompensatedValues": false,
  "model": "I-9PSL",
  "httpDomain": "",
  "disableCloudConnection": false
}
```

is this the expected behavior? See the video below.

https://github.com/user-attachments/assets/6d010791-2c86-49f7-936a-9bba38a753d1

ccoley, 2025-04-30:
I can confirm this issue as well. I have 4 monitors on firmware v3.3.6 and a prometheus instance pulling metrics from the `/metrics` endpoint regularly. All of the monitors have been in a reset loop for about a day, maybe longer.

I downgraded one the monitors to v3.1.13 manually, and that resolved the issue.
I then upgraded that monitor to v3.2.0 and the issue is still resolved.
If I stop prometheus from polling the monitors, that also resolves the issue.

ccoley, 2025-05-01:
I see this error in the console when the reset happens on v3.3.6:

```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4202d16e  RA      : 0x42004aa2  SP      : 0x3fcb34f0  GP      : 0x3fc8e800
TP      : 0x3fc6e56c  T0      : 0x61204550  T1      : 0x6b6f5f67  T2      : 0x59542023
S0/FP   : 0x3fc920dc  S1      : 0x3c14f000  A0      : 0x00000000  A1      : 0x00000000
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000080
A6      : 0x000a6567  A7      : 0x75616720  S2      : 0x3c159000  S3      : 0x3c133000
S4      : 0x4200bf64  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x3fcb35dc
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x69666e6f  T4      : 0x635f746e  T5      : 0x65696461  T6      : 0x72677269
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000041
MHARTID : 0x00000000

Stack memory:
3fcb34f0: 0x3fc9312c 0x00000000 0x00000000 0x4206c31c 0x3fcb40e8 0x3fcb46b8 0x3fc91f8c 0x3fcb351c
3fcb3510: 0x3fcb35dc 0x3fc
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4204b750
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
Serial nr: [REDACTED]
Reset reason: ESP_RST_TASK_WDT
```

I manually compiled version v3.2.0 of the firmware and it works as expected. I then manually compiled v3.3.0 of the firmware, it compiles and uploads fine, but polling the `/metrics` endpoint causes the reset again. It seems that this error was introduced in v3.3.0

samuelbles07, 2025-05-01:
Hi @ccoley , thank you for pointing this out, and you're right to spot the problem. If its fine for you, could you try this fix #310 ?

ccoley, 2025-05-01:
@samuelbles07 The fix in #310 does resolve this issue.

closed 2025-05-04 by samuelbles07