Open-Air-Foundation/firmware-one-openair #312: AirGradient is in boot loop after disabling its network access

szotsaki, 2025-05-01:
I flashed the firmware to a new version (3.3.6), made a manual factory reset, and connected to WiFi. I disabled its network access.

Since then the device is in boot loop:

```
[Configure] Info: Reading file...
[Configure] Info: Reading file: success
[Configure] Info: Parse file to JSON
[Configure] Info: corrections not found
[Configure] Info: {"country":"TH","mqttBrokerUrl":"","httpDomain":"","configurationControl":"both","pmStandard":"ugm3","temperatureUnit":"c","disableCloudConnection":true,"postDataToAirGradient":true,"ledBarBrightness":100,"displayBrightness":100,"ledBarMode":"co2","tvocLearningOffset":12,"noxLearningOffset":12,"abcDays":8,"model":"","offlineMode":false,"monitorDisplayCompensatedValues":false}
Detected ONE_INDOOR
[OledDisplay] Info: begin
Firmware Version: 3.3.6
LED bar brightness: 100
Init SGP41 success
initializing PM sensor
cleared 0 byte(s)
setting active mode
7 byte(s) written
Set S8 AbcDays successful
[AGSERIAL] Info: IICSerial initialized
[AGSERIAL] Warning: IICSerial failed open serial line, retry..
[AGSERIAL] Warning: IICSerial failed open serial line, retry..
[AGSERIAL] Warning: IICSerial failed open serial line, retry..
[AGSERIAL] Error: IICSerial failed open serial line, give up..
Cellular module not available, using wifi
*wm:AutoConnect 
*wm:Connecting to SAVED AP: <AP SSID>
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address:
[WifiConnector] Info: Wait for configure portal
[WifiConnector] Info: WiFi Connected: <AP SSID> IP: 0.0.0.0
[LocalServer] Info: Init: airgradient_744dxxxxxxxxxx.local
MQTT is not configured, skipping initialization of MQTT client
Display brightness: 100
Success create networking task
Running monitor without connection to AirGradient server
E (23165) esp-tls: [sock=49] select() timeout
E (23165) TRANSPORT_BASE: Failed to open a new connection: 32774
E (23166) HTTP_CLIENT: Connection failed, sock < 0
[OTAWifi] Error: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT

assert failed: _prvTaskExitError port.c:126 (uxCriticalNesting == ~0UL)
Core  0 register dump:
MEPC    : 0x40382902  RA      : 0x40387232  SP      : 0x3fcb4320  GP      : 0x3fc8e800  
TP      : 0x3fc6f2dc  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x3fcb4497  S1      : 0x00000080  A0      : 0x3fcb4384  A1      : 0x3fc91361  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc9a000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcb4378  S3      : 0x00000001  
S4      : 0x3fcb4378  S5      : 0x40389ec6  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fcb4320: 0x6e6f6320 0x7463656e 0x3c14e460 0x4038cb00 0x5252455f 0x00000000 0x3fc99848 0x3fc91360
3fcb4340: 0x3fc8e0b8 0x3fc99850 0x000000ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4204b750
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
Serial nr: 744dxxxxxxxxxx
Reset reason: ESP_RST_TASK_WDT
```
```

szotsaki, 2025-05-02:
- Not an offline mode in the sense that I don't push the button during boot because I need local access
- Ticked the "offline mode" (I forgot what it's called exactly) checkbox when I was setting up the WiFi access which assured that the update searches and other internet communication won't happen 
- Restricted its internet access within my router but local access is still enabled

samuelbles07, 2025-05-04:
Ok, so basically the checkbox on wifi portal is disabling every communication to airgradient server. Do you happen using Prometheus on your local setup? Because there's an issue relating this #309 .

szotsaki, 2025-05-04:
I use Home Assistant (not prometheus) but I don't know if it reaches that particular endpoint. However, I see that the exception is different, not thrown from the same place, therefore I think it's a different issue.

samuelbles07, 2025-05-05:
Hi @szotsaki , you're definitely right. Is it possible to you try this PR? #313

samuelbles07, 2025-05-07:
PR #313 already merged. And it should fix this issue, closing this.

closed 2025-05-07 by samuelbles07