Open-Air-Foundation/firmware-one-openair #315: Connecting to wifi broken since 3.3.6

sheld0r73, 2025-05-10:
My airgradient one updated last night to firmware 3.3.6
Since then it won't connect to my wifi

For testing purposes I've installed 3.0.8 directly from [airgradient](https://www.airgradient.com/documentation/firmwares/) which is working perfectly fine
However this old version doesn't support home assistant

Below is the log from firmware 3.3.6
I don't understand why it assignes the IP 0.0.0.0
I'm providing the airgradient with a fixed IP and as said everything is working fine with 3.0.8

I can flash between those two versions as often as I like and the result will be always the same
3.0.8 is working but 3.3.6 doesn't connect to the wifi

```
OTA progress: 94
OTA progress: 97
OTA progress: 99
OTA progress: 100
[OTAWifi] Info: Download iamge binary complete, applying image...
[OTA] Info: Finishing... Total binary data length written: 1727616
[OTA] Info: OTA successful, MAKE SURE TO REBOOT!
OTA update performed, restarting ...
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4201caa8  RA      : 0x420144ea  SP      : 0x3fcb5590  GP      : 0x3fc8e800  
TP      : 0x3fc705fc  T0      : 0x00000800  T1      : 0x40389898  T2      : 0x0000000e  
S0/FP   : 0x3fc920b0  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x00000000  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x600c0000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x60013000  S2      : 0x3fc99000  S3      : 0x3fc9a000  
S4      : 0x3fc997ac  S5      : 0x3c132000  S6      : 0x3fc92000  S7      : 0x3c13e000  
S8      : 0x4211f000  S9      : 0x0000aa53  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000078  T4      : 0x00000001  T5      : 0x3fcb5390  T6      : 0x00000001  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000230  
MHARTID : 0x00000000  

Stack memory:
3fcb5590: 0x3fc99000 0x3fc93000 0x3fc92000 0x420008e6 0x00000000 0x0000aa53 0x4211f000 0x00000000
3fcb55b0: 0x00000000 0x000ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4204b754
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
Serial nr: XXXserialXXX
Reset reason: ESP_RST_TASK_WDT
[Configure] Info: Reading file...
[Configure] Info: Reading file: success
[Configure] Info: Parse file to JSON
[Configure] Info: corrections not found
[Configure] Info: {"country":"TH","pmStandard":"ugm3","ledBarMode":"co2","abcDays":8,"tvocLearningOffset":12,"noxLearningOffset":12,"mqttBrokerUrl":"","httpDomain":"","temperatureUnit":"c","disableCloudConnection":false,"configurationControl":"both","postDataToAirGradient":true,"ledBarBrightness":100,"displayBrightness":100,"offlineMode":false,"monitorDisplayCompensatedValues":false}
Detected ONE_INDOOR
[OledDisplay] Info: begin
Firmware Version: 3.3.6
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x42080086
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
Serial nr: XXXserialXXX
Reset reason: ESP_RST_UNKNOWN
[Configure] Info: Reading file...
[Configure] Info: Reading file: success
[Configure] Info: Parse file to JSON
[Configure] Info: corrections not found
[Configure] Info: {"country":"TH","pmStandard":"ugm3","ledBarMode":"co2","abcDays":8,"tvocLearningOffset":12,"noxLearningOffset":12,"mqttBrokerUrl":"","httpDomain":"","temperatureUnit":"c","disableCloudConnection":false,"configurationControl":"both","postDataToAirGradient":true,"ledBarBrightness":100,"displayBrightness":100,"offlineMode":false,"monitorDisplayCompensatedValues":false}
Detected ONE_INDOOR
[OledDisplay] Info: begin
Firmware Version: 3.3.6
LED bar brightness: 100
Init SGP41 success
initializing PM sensor
cleared 0 byte(s)
setting active mode
7 byte(s) written
Set S8 AbcDays successful
[AGSERIAL] Info: IICSerial initialized
[AGSERIAL] Warning: IICSerial failed open serial line, retry..
[AGSERIAL] Warning: IICSerial failed open serial line, retry..
[AGSERIAL] Warning: IICSerial failed open serial line, retry..
[AGSERIAL] Error: IICSerial failed open serial line, give up..
Cellular module not available, using wifi
*wm:AutoConnect 
*wm:Connecting to SAVED AP: MyWiFi24
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address:
[WifiConnector] Info: Wait for configure portal
**[WifiConnector] Info: WiFi Connected: MyWiFi24 IP: 0.0.0.0**
[LocalServer] Info: Init: airgradient_XXXserialXXX.local
MQTT is not configured, skipping initialization of MQTT client
[AgWifiClient] Info: Post measures to https://hw.airgradient.com/sensors/airgradient:XXXserialXXX/measures
[AgWifiClient] Error: Failed post measures to server with response code 400
[AgWifiClient] Info: Fetch configuration from https://hw.airgradient.com/sensors/airgradient:XXXserialXXX/one/config
[AgWifiClient] Error: Failed fetch configuration from server with return code 400
Display brightness: 100
Success create networking task
Firmware update starting...

```

p.s. the airgradient is stuck in a loop with 3.3.6
once it's done with the firmware update it'll reboot and try again

okica-11, 2025-05-10:
I'm having exactly the same problems since last night's auto-update. 
Same as you, I can stop this problem by installing the older version 3.0.8. 

Check #308 related to #309. Honestly, I'm not very happy with how much all this reinstalling will break my home assistant integration.

Iroxxar, 2025-05-10:
As @okica-11 I had the same Issue since last night. Had to flash to 3.2.0 via Arduino-IDE. 

In my opinion Flashing 3.0.8 is not a solution, since features are missing and bugs are present in the prometheus metric standard. 

I'd suggest providing 3.2.0 as a "quick-flash" the [firmwares](https://www.airgradient.com/documentation/firmwares/) section.

This might still be a problem, since the device tries to OTA-update itself with a normal firmware, therfore the firmware has to be set with the disabled ota.

The general error in the current 3.3.6 seems to be with:

| `[WebServer.cpp] _handleRequest(): request handler not found`

okica-11, 2025-05-10:
Chiming in again so fast, lol. 

Anyway, I tried to install 3.2.0 but decided there must be a faster way (installing the update via a web browser). 

Being unable to find 3.2.0 web-updater, I did find version 3.1.13 here https://www.airgradient.com/documentation/one-v9/ . So I wiped and installed the old FW via USB and browser and connected to the AP. I entered local wifi credentials and DISABLED OTA with a checkmark at the bottom of the page. Soon enough air sensor somehow decides it still wants to do the OTA update (even when it should not!) and we are again, back to square one with self-rebooting-not-working-3.3.6-forcing air sensor.

I give up. I hope they push _working_ OTA update soon. Until then, my sensor is unplugged and waiting in the corner - measuring dust particles the old way ;).

alexhk90, 2025-05-10:
> I'm also seeing a 3.3.6 firmware update loop, but wifi works fine. My solution has been to intercept DNS requests and ban hw.airgradient.com for now.
> 
> My understanding is that the OTA request to `http://hw.airgradient.com/...` should return 304 if firmware is up to date. However, the server sends 3.3.6 firmware again. This better be fixed soon as the loop will cause flash degradation.

Thanks for posting, @sh0 ! Blocking DNS requests from my AirGradient for "hw.airgradient.com" has stopped the 3.3.6 boot loop that started this morning for me at around 4am LDN (3am UTC), so it's usable again until the fixed version of the firmware is rolled out. 

This does make me wonder if automatic firmware updates like this should be enabled by default - I don't remember enabling them but perhaps there was an option to disable them that I missed. Maybe it would be better for automatic firmware updates to be disabled by default (with an option to enable) but have it show as available via the HA integration.

sheld0r73, 2025-05-10:
> I'm also seeing a 3.3.6 firmware update loop, but wifi works fine. My solution has been to intercept DNS requests and ban hw.airgradient.com for now.
>
> My understanding is that the OTA request to [...]

Thx @sh0 - Blocking DNS requests to hw.airgradient.com is a decent workaround

I hope they'll release a fix soon...

samuelbles07, 2025-05-12:
Hi All,

We just release 3.3.7 to fix this, but its not on the firmware page and automatic firmware update yet. For more information about this issue, you can read this explanation: https://forum.airgradient.com/t/all-airgradient-one-sensors-repeatedly-updating-on-3-3-6-starting-09-30-gmt-29-04/4069/37?u=samuel_airgradient

DyonR, 2025-05-13:
I've updated to 3.3.7 and I am not getting any bootloops anymore 👍🏻

alexhk90, 2025-05-17:
To confirm, I've also now updated to 3.3.7 which has fixed the issue.

okica-11, 2025-06-16:
Somehow, the fix still doesn't work for me. When I update to version 3.3.8 via the web flasher, the device connects to Wi-Fi and installs version 3.3.7 (this is also mentioned in the comment section of the forum linked above)! The symptoms are the same as with the broken version - restarts.

To be honest, I'm frustrated because: 
- this problem has been ongoing for almost a month now
- version 3.3.8 becomes 3.3.7 and 
- to add even more salt to the wound, I'm running the sensor **locally** with the idea that I would have full control over updates, too.

toxema5332, 2025-06-16:
So which was the latest stable version for you? 3.1.13?
Have you considered downgrading?

If so then you can easily downgrade to 3.1.13 by modifying the firmware version in flashing web page.

Just go to:
https://www.airgradient.com/documentation/firmwares/

Open the DevTools (F12), go to the Elements tab, search for this:
`/documents/fw/new/combo/3-3-8/manifest.json?_cache_buster=1136`
Replace it with this:
`/documents/fw/new/combo/3-1-13/manifest.json`
Then click the "Flash Now" button (next to "Version 3.3.8 Release")

Cheers

okica-11, 2025-06-25:
Sorry for the late reply. I'm not 100% sure which version was still stable for me - whichever was automatically pushed via OTA around the start of May 2025 (before the problem caused by 3.3.6).

I'm not keen on downgrading from 3.3.6 by so many versions to 3.1.13. Looking back at my comments, that wouldn't work anyway. The 3.0.8 did work, but 3.1.13 auto-updated itself.

Anyway, feeling frustrated, I decided to block hw.airgradient.com on the router as @sh0 suggested at the beginning of the thread. It seems to be working now...

> I'm having exactly the same problems since last night's auto-update. Same as you, I can stop this problem by installing the older version 3.0.8.
>
> Check [...]

> Chiming in again so fast, lol.
>
> Anyway, I tried to install 3.2.0 but decided there must be a faster way (installing the update via a web browser).
>
> Being unable to find 3.2.0 web-updater, I did find [...]