Open-Air-Foundation/firmware-one-openair #320: CO2 Readings Stop Until Hard Restart (S8 Sensor Shutdown)

macserv, 2025-05-18:
## Problem

I have two AirGradientOne I-9PSL units (pre-soldered kits, purchased at the same time).  Both devices, after an indeterminate length of time, will stop reporting a CO2 reading until restarted with a full power cycle.

Note: I'm new to embedded device programming, but not to software development in general, so if someone can walk me through how to debug the code running on the device (or point me to a good tutorial), I'd be happy to dig deeper.  A quick search suggests I might need additional (JTAG) hardware, but if it can be done over the USB connection, I'll give it a shot.

## Troubleshooting Notes

- Captured log output in several scenarios... see "Serial Monitor Output" below. 
- Tried several USB cables and power adapters (including the direct connection to computer while monitoring) with no discernible change to behavior.
- A soft reset is not sufficient to restore normal operation; the power must be cycled (i.e., disconnect and reconnect the USB cable).

## Serial Monitor Output

### Normal Operation

When functioning normally, the serial monitor output shows valid responses from the S8.

```
[125681][I][S8.cpp:757] validResponse(): Valid response
[125752][I][S8.cpp:244] getCo2(): CO2 value = 866 ppm
[125926][I][Sgp41.cpp:339] setCompensationTemperatureHumidity(): Update: defaultT: 25367, defaultRh: 31480

CO2 = 866.00 ppm
Temperature = 22.74 C
Relative Humidity = 48.06
...
```

### Malfunction

When the device stops reporting CO2 readings, the serial monitor output indicates invalid responses from the S8.

```
[23941108][I][S8.cpp:786] validResponseLenght(): Err: Unexpected length!
[23941196][I][S8.cpp:339] getCalibPeriodABC(): Error getting ABC period!
[23942459][I][S8.cpp:786] validResponseLenght(): Err: Unexpected length!
[23942547][I][S8.cpp:246] getCo2(): Error getting CO2 value!
CO2{1} invalid value update counter reached (3x)! Setting its average value to invalid!
[23942728][I][Sgp41.cpp:339] setCompensationTemperatureHumidity(): Update: defaultT: 24771, defaultRh: 32197
Last process timeout, clear buffer and last handle package

CO2 = -
Temperature = 21.16 C
Relative Humidity = 49.13
...
```

### Warm Reset

If I use the PlatformIO "Upload and Monitor" task in to restart the device without cycling the power, the monitor output indicates that the S8 is not found.

```
[  7297][I][S8.cpp:659] init(): Init on 'Serialx'
[  7302][V][esp32-hal-uart.c:330] uartBegin(): UART1 baud(9600) Mode(800001c) rxPin(0) txPin(1)
[  7312][V][esp32-hal-uart.c:416] uartBegin(): UART1 not installed. Starting installation
[  7321][V][esp32-hal-uart.c:467] uartBegin(): UART1 initialization done.
[ 12437][I][S8.cpp:786] validResponseLenght(): Err: Unexpected length!
[ 12445][I][S8.cpp:96] getFirmwareVersion(): Firmware version not available!
CO2 S8 sensor not found
...
[StateMachine] Warning: CO2 S8 not available, set 'abcDays' ignored
HTTP domain name from configuration empty, set to default
[ 54468][I][Sgp41.cpp:339] setCompensationTemperatureHumidity(): Update: defaultT: 25108, defaultRh: 32959

Temperature = 22.03 C
Relative Humidity = 50.30
```

### Cold Reset

If I disconnect and reconnect the USB cable, the S8 begins functioning again:

```
[  7302][I][S8.cpp:659] init(): Init on 'Serialx'
[  7307][V][esp32-hal-uart.c:330] uartBegin(): UART1 baud(9600) Mode(800001c) rxPin(0) txPin(1)
[  7317][V][esp32-hal-uart.c:416] uartBegin(): UART1 not installed. Starting installation
[  7326][V][esp32-hal-uart.c:467] uartBegin(): UART1 initialization done.
[  7455][I][S8.cpp:757] validResponse(): Valid response
[  7461][I][S8.cpp:94] getFirmwareVersion(): Firmware version: 1.2
[  7468][I][S8.cpp:679] init(): Firmware version: 1.2
[  7473][I][S8.cpp:681] init(): Sensor successfully initialized. Heating up for 10s
...
[  7964][I][S8.cpp:757] validResponse(): Valid response
[  8035][I][S8.cpp:337] getCalibPeriodABC(): ABC period: 192 hour
Set S8 AbcDays successful
...
[ 44910][I][S8.cpp:757] validResponse(): Valid response
[ 44981][I][S8.cpp:244] getCo2(): CO2 value = 1257 ppm
[ 47221][I][Sgp41.cpp:339] setCompensationTemperatureHumidity(): Update: defaultT: 25214, defaultRh: 32860

CO2 = 1257.50 ppm
Temperature = 22.33 C
Relative Humidity = 50.20
...
```

samuelbles07, 2025-05-19:
Hi @macserv ,

Do you still have this problem after the cold reset? If not, perhaps its relate to this this forum https://forum.airgradient.com/t/all-airgradient-one-sensors-repeatedly-updating-on-3-3-6-starting-09-30-gmt-29-04/4069

macserv, 2025-05-19:
Hey, @samuelbles07 ,

Good question!  Yes, the issue will always recur.  To clarify, I've updated the troubleshooting notes above with the following item:

> After cycling the power, the CO2 sensor will function normally again for an indeterminate length of time, but the issue will always recur.

macserv, 2025-07-14:
@samuelbles07 @pnt325 @airgradienthq Hey, all, following up after a couple of months... can anyone provide some guidance?  Thanks!

jsjhlllx, 2026-10-09:
Following this issue while designing an S8-based board, so two questions for @macserv and @samuelbles07:

From the logs, the key detail is that a warm reset of the ESP32 (S8 stays powered) still gets "Unexpected length", while a full power cycle recovers immediately. That points at the S8 itself hanging rather than the ESP UART driver.

1. @macserv — when it's in the stuck state, have you been able to measure the S8's 5 V supply (G+ pin) and whether the S8 still emits anything on its TX line? Even a multimeter reading of the rail would help separate "S8 firmware hang" from "brown-out on the 5 V rail".
2. @samuelbles07 — has AirGradient seen this across many units / S8 firmware versions (this one is FW 1.2), and has Senseair commented?

For a new design I'm planning to put the S8's 5 V behind an MCU-controlled load switch so firmware can power-cycle it after N consecutive invalid responses. If that matches what you've observed, it might be worth doing on a future ONE revision too.