Open-Air-Foundation/firmware-one-openair #329: Local configuration has 2 identical temperatureUnit field name

samuelbles07, 2025-06-18:
## Problem

As mentioned in [this issue](https://github.com/home-assistant/core/issues/144039), there appears to be a case where the same field name, `temperatureUnit`, exists twice in the JSON payload. This results in one of the `temperatureUnit` fields having a `null` value, which causes the [`python-airgradient`](https://github.com/airgradienthq/python-airgradient) library to crash when attempting to parse the configuration.

Here is the problematic JSON payload:

```json
{
  "country": "US",
  "pmStandard": "us-aqi",
  "ledBarMode": "pm",
  "abcDays": 8,
  "tvocLearningOffset": 12,
  "noxLearningOffset": 12,
  "mqttBrokerUrl": "",
  "temperatureUnit": "f",
  "configurationControl": "cloud",
  "postDataToAirGradient": true,
  "ledBarBrightness": 10,
  "displayBrightness": 20,
  "offlineMode": false,
  "model": "I-9PSL",
  "monitorDisplayCompensatedValues": false,
  "disableCloudConnection": false,
  "corrections": {
    "atmp": {
      "correctionAlgorithm": "none",
      "slr": null
    },
    "rhum": {
      "correctionAlgorithm": "none",
      "slr": null
    }
  },
  "httpDomain": "",
  "temperatureUnit": null
}
```

Even though updating `temperatureUnit` locally appears to succeed, when the configuration is retrieved again, the change isn't reflected.

Interestingly, the [ArduinoJSON](https://github.com/airgradienthq/arduino/tree/master/src/Libraries/Arduino_JSON) library is able to parse this JSON without issue, which may contribute to the inconsistency.

## Temporary solution

There's a quick way to fix this to just factory reset the monitor. But there's a possibility this problem could still occur in the future if the core problem is not fixed.

GuySie, 2025-06-18:
Ofcourse, I've sent you the logs. The changes set don't seem to matter, I still can't add to HA and if I check the config it looks like temperatureUnit is still null:

```{"country":"NL","pmStandard":"ugm3","ledBarMode":"co2","abcDays":8,"tvocLearningOffset":12,"noxLearningOffset":12,"mqttBrokerUrl":"","temperatureUnit":"c","configurationControl":"local","postDataToAirGradient":true,"ledBarBrightness":10,"displayBrightness":20,"offlineMode":false,"monitorDisplayCompensatedValues":false,"model":"I-9PSL-DE","corrections":{"pm02":{"correctionAlgorithm":"slr_PMS5003_20240104","slr":{"intercept":0,"scalingFactor":0.02896,"useEpa2021":true}},"atmp":{"correctionAlgorithm":"none","slr":null},"rhum":{"correctionAlgorithm":"none","slr":null}},"disableCloudConnection":false,"httpDomain":"","temperatureUnit":null}```

samuelbles07, 2025-06-19:
Got it. So the problem here how is it possible the json has two `temperatureUnit` field.

Could you try to change the `temperatureUnit` value to "f" @GuySie ?

```
curl -X PUT -H "Content-Type: application/json" -d '{"temperatureUnit":"f"}' http://airgradient_<SERIAL_NUMBER>.local/config
```

GuySie, 2025-06-19:
@samuelbles07 Just sent you the new logs. Still no joy getting it back into Home Assistant. Config is now:

```{"country":"NL","pmStandard":"ugm3","ledBarMode":"co2","abcDays":8,"tvocLearningOffset":12,"noxLearningOffset":12,"mqttBrokerUrl":"","temperatureUnit":"f","configurationControl":"local","postDataToAirGradient":true,"ledBarBrightness":10,"displayBrightness":20,"offlineMode":false,"monitorDisplayCompensatedValues":false,"model":"I-9PSL-DE","corrections":{"pm02":{"correctionAlgorithm":"slr_PMS5003_20240104","slr":{"intercept":0,"scalingFactor":0.02896,"useEpa2021":true}},"atmp":{"correctionAlgorithm":"none","slr":null},"rhum":{"correctionAlgorithm":"none","slr":null}},"disableCloudConnection":false,"httpDomain":"","temperatureUnit":null}```

samuelbles07, 2025-06-20:
Thanks for this @GuySie , maybe one more question, is your monitor oled display for temperature unit changed from Celcius to Farenheit?

GuySie, 2025-06-20:
Hi @samuelbles07 yes it is now set to F on the display.

GuySie, 2025-07-10:
Anything I can still help you with @samuelbles07 , or is there a fix you need me to test in the near future? If not, I will factory reset my unit so I can use it in HA again.