Open-Air-Foundation/firmware-one-openair #335: Unable to flash AirGradient One V9 (indoor) from source

mtlynch, 2025-08-10:
I recently purchased a pre-assembled AirGradient One V9, but I'm unable to flash source onto the device. If I compile the latest tagged release, it says that the compiled code exceeds the board's space.

```bash
$ git checkout 3.3.9 && git submodule update --recursive
Previous HEAD position was c841476 Merge pull request #247 from airgradienthq/fix/pms-read-data
HEAD is now at d8eb6b3 Prepare release 3.3.9
Submodule path 'src/Libraries/airgradient-client': checked out 'a4ac14936e91bca7a660536293f2e52297798875'
Submodule path 'src/Libraries/airgradient-ota': checked out 'c772392427cf343cbb9afedfb395e5a027cfb046'
```

```bash
$ arduino-cli compile \
    --verbose \
    --fqbn esp32:esp32:lolin_c3_mini \
    --library . \
    --port /dev/ttyACM0 \
    --verify \
    --upload \
    examples/OneOpenAir/OneOpenAir.ino
...
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Sketch uses 1560758 bytes (119%) of program storage space. Maximum is 1310720 bytes.
Global variables use 51060 bytes (15%) of dynamic memory, leaving 276620 bytes for local variables. Maximum is 327680 bytes.

Used library                   Version Path
AirGradient Air Quality Sensor 3.3.9   /home/mike/airgradient-one
Wire                           2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/Wire
EEPROM                         2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/EEPROM
ESPmDNS                        2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/ESPmDNS
WebServer                      2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/WebServer
WiFi                           2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/WiFi
HTTPClient                     2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/HTTPClient
WiFiClientSecure               2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/WiFiClientSecure
FS                             2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/FS
SPIFFS                         2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/SPIFFS
Update                         2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/Update
DNSServer                      2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/DNSServer
SPI                            2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/SPI

Used platform Version Path
esp32:esp32   2.0.17  /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17
Error during build: text section exceeds available space in board
```

The latest version I could get to flash successfully is `3.1.9`, but then it just boot loops:

```bash
$ arduino-cli compile \
    --verbose \
    --fqbn esp32:esp32:lolin_c3_mini \
    --library . \
    --port /dev/ttyACM0 \
    --verify \
    --upload \
    examples/OneOpenAir/OneOpenAir.ino
...
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: d8:3b:da:1a:ee:c4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
...
Writing at 0x0015d845... (100 %)
Wrote 1371712 bytes (806220 compressed) at 0x00010000 in 12.0 seconds (effective 916.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Used library                   Version Path
AirGradient Air Quality Sensor 3.1.9   /home/mike/airgradient-one
Wire                           2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/Wire
EEPROM                         2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/EEPROM
ESPmDNS                        2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/ESPmDNS
WebServer                      2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/WebServer
WiFi                           2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/WiFi
HTTPClient                     2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/HTTPClient
WiFiClientSecure               2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/WiFiClientSecure
FS                             2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/FS
SPIFFS                         2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/SPIFFS
Update                         2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/Update
DNSServer                      2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/DNSServer
SPI                            2.0.0   /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17/libraries/SPI

Used platform Version Path
esp32:esp32   2.0.17  /home/mike/.arduino15/packages/esp32/hardware/esp32/2.0.17
```

```bash
$ arduino-cli monitor --port /dev/ttyACM0
Using default monitor configuration for board: esp32:esp32:heltec_wifi_lora_32_V3
Monitor port settings:
  baudrate=9600
  bits=8
  dtr=on
  parity=none
  rts=on
  stop_bits=1

Connecting to /dev/ttyACM0. Press CTRL-C to exit.
ESP-ROM:esp32c3-api1-20210207
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf94c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf94c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x2624
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
```

How can I compile a working version from source? This is one of the flagship features advertised for the device, and it doesn't appear to work.

mtlynch, 2025-08-13:
@samuelbles07 - Thanks! I've updated the README in #336 to make this discoverable for users in the future.

samuelbles07, 2025-08-13:
Glad it help @mtlynch ! Thank you for the contribution!

PR #336 (2025-08-13): Link to inner docs pages from main README
Fixes #335

There's useful documentation in the docs/ directory, but it's hard for users to discover it, as there are no links to it.

This change updates the README to add links to the inner docs pages for users interested in customizing and re-flashing firmware onto their devices as well as understanding the OTA update system.
closed 2025-08-13 by samuelbles07